SignalK/SensESP #16: Doesn't reconnect to wifi if wifi goes off and comes back on

ba58smith, 2019-05-30:
I thought this was happening, so I just restarted my wifi to test it, and it is.

Turned wifi on, turned D1 Mini on, SensESP connected and started sending data. Was sending data for a few hours. I turned off the wifi, and SensESP immediately detected that, and restarted. When it came up, it didn't find my wifi (it was still off), so it went into AP mode as it's supposed to. But then, after I turned the wifi on, and SensESP was seeing the wifi, it still didn't connect to it. I had to go to Unconfigured Sensor and re-select my wifi and re-enter the password, and then it was OK. Here's the serial output of the whole test, starting with me turning the wifi off.
```
␛[0m(E) (check_connection) Wifi disconnected: restarting...
␛[0m
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.systemhz
␛[0m(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.freemem
␛[0m(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.uptime
␛[0m(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.ipaddr
␛[0m(I) (enable) Connecting SignalK source propulsion.left.revolutions
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (enable) Subsystem: networking->setup()
␛[0m*WM: Adding parameter
*WM: hostname
*WM:
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 1
*WM: SET AP STA
*WM:
*WM: Configuring access point...
*WM: Unconfigured Sensor
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
*WM: About to scan()
*WM: Scan done 
// it does this about 20 times.
// I turned my router back on about here....
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
(E) (setup) Failed to connect to wifi and config timed out. Restarting...
␛[0m
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.systemhz
␛[0m(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.freemem
␛[0m(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.uptime
␛[0m(I) (enable) Connecting SignalK source sensors.WemosD1miniTest.ipaddr
␛[0m(I) (enable) Connecting SignalK source propulsion.left.revolutions
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (enable) Subsystem: networking->setup()
␛[0m*WM: Adding parameter
*WM: hostname
*WM:
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result:
*WM: 0
*WM: SET AP STA
*WM:
*WM: Configuring access point...
*WM: Unconfigured Sensor
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: theturtle_IoT
*WM: DUP AP: theturtle
*WM: DUP AP: fos_guest
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: fos_guest
*WM: DUP AP: theturtle
*WM: DUP AP: theturtle_IoT
// So it's obviously seeing the network, but won't connect to it.
```
It's now more than an hour after I pasted the above in, and I'm just finishing this. It's still doing what it was doing above: scanning, finding the networks, but not connecting to them. So I just restarted the Wemos, and it's doing the same thing. I have to go into Unconfigured Sensor and tell it my wifi info again, and then it's OK.

joelkoz, 2019-05-31:
Line 51 of net/networking.cpp calls the wifi manager's autoConnect() method to connect to wifi.  In that call, there is an optional parameter to specify the number of retries before falling back into access point mode.  The default number of retries is 1.  So, that is the reason for the current behavior.

I'm thinking we should have user configurable option that would allow the max retries to be set.  A max retry of "1" should ALWAYS be used if the sensor has NEVER been configured before.  If it HAS been configured before, the the user should be able to set the number of retries, and the delay between each retry (both are parameters to that call). 

Of course, we need to be careful about what happens if the user wants to change the Wifi on purpose, or for whatever reason, loses the old setting permanently.  If the gauge is configured to continously retry, you'd lose communication with it all together.

Maybe a better approach is to retry a certain number of times, and if that fails, go into AP mode, but if no configuration occurs in X number of minutes, the system restarts and tries to retry again?  I think the SensESP configuration IS in fact saved - its just that the wifi manager goes into AP mode upon first failure and then just waits there.

joelkoz, 2019-05-31:
Upon further review...

It appears that the behavior of "if no AP config occurs in X number of minutes, reboot and retry" is already implemented.

Line 42 of networking.cpp has this:

`wifi_manager->setConfigPortalTimeout(WIFI_CONFIG_PORTAL_TIMEOUT)`

and WIFI_CONFIG_PORTAL_TIMEOUT is currently set at 180 seconds.

So, it does appear that somewhere in this process, SensESP is losing the previous configuration. If it had kept it, after the 180 second timeout, it should have reconnected.

ba58smith, 2019-05-31:
Exactly. I'll try what you suggested (using a debug message to display the SSID just before it tries to connect), and see if that reveals anything.

ba58smith, 2019-06-13:
I added some debug messages in networking.cpp, and once the ESP was up and running properly, I turned the wifi off and back on. The serial output is in the attached text file. The two problems seem to be:
1. Even after the wifi network becomes visible again in the scan, ESP doesn't connect to it, and eventually times out and restarts. This could be my network - on my own computer, I can see the network SSID, but it takes my computer several attempts to finally connect to it. (The router takes a long time to fully boot up.)
2. After the first timeout-and-start over, the network SSID is no longer known, so it will never reconnect at that point.
I'm going to make my WIFI_CONFIG_PORTAL_TIMEOUT really long - 5 minutes - and see if problem #1 isn't really a SensESP problem. But problem #2 certainly seems to be, unless I'm trying to display a value that hasn't been set it memory yet because the wifi hasn't restarted. Maybe my debug message needs to be pulling directly from the configuration file stored on the device?
[SensESP loss of network issue.txt](https://github.com/SignalK/SensESP/files/3286175/SensESP.loss.of.network.issue.txt)

ba58smith, 2019-06-13:
I set my WIFI_CONFIG_PORTAL_TIMEOUT to 420 (seven minutes) and got the same problem in #1, above - it knows the SSID it's supposed to connect to, and it sees the network, but it never connects.

ba58smith, 2019-06-13:
I added some more debug messages, and now I can see (in the attached file) that the wifi SSID is still known by SensESP before the ESP.restart(), but I'm guessing the wifi SSID is somehow lost during the ESP.restart(), because although the network is found, it doesn't connect, and when it times out, just before it restarts again, it no longer knows the wifi SSID.

At this point, I don't know what to try next.

[SensESP loss of network issue2.txt](https://github.com/SignalK/SensESP/files/3286566/SensESP.loss.of.network.issue2.txt)

PR #53 (2019-06-26): fix: Hard-code network settings after one successful connection
With this one, I'd like you to think hard whether you like my approach. Basically, I found that the bug @ba58smith reported was almost a show-stopper for me. I had some sensors behind the same power switch as the SK server itself. When I turned the power on, the Raspi couldn't get the network up fast enough and the ESPs would fall back to unconfigured mode. Every single time.

What I do is I save SSIDs and network passwords after a successful network login via Wifimanager. Then, if the values have been set, they're used unconditio [... cut]
closed 2019-06-26 by mairas