SignalK/SensESP #57: ESP32

andyrbarrow, 2019-07-20:
Do you have an example that compiles with ESP32? I can't get any of the examples to compile, even after changing platformio.ini.

joelkoz, 2019-07-20:
I don't at the moment, but I just received my ESP32 in the mail today for my next project.  I'll give it a go and see what happens.

andyrbarrow, 2019-07-20:
Thanks. It appears the issue is with ESP8266Random. I've been trying to
work my way through the code to figure out opportunities for porting to the
ESP32.
*Andy Barrow*

*email: *andy@sailor.nu
*tel (mx):* +52 322 104 2189
*tel (us):* +1 925 270 0473
*Skype:* abarrow

andyrbarrow, 2019-07-20:
I fiddled with it a bit, but I can't get it to work. I took out ESP8266Random and replaced the RANDOM_REG32 definition with a variable that gets a random number from esp_random(). It fixed a few things, but there are lots of other calls to 8266 specific libraries.

joelkoz, 2019-07-20:
My guess is that fairly high up the chain, a compiler directive needs to be used in the SensESP code to select an ESP32 version of a particular library (vs. an ESP8266).  For example:

#ifdef ESP8266
  #include <ESP8266mDNS.h> // Include the mDNS library
#elif defined(ESP32)
  #include <ESPmDNS.h>
#endif

andyrbarrow, 2019-07-20:
Yes, that's what I was doing. I'll keep hacking away at it.

Haimana, 2020-02-15:
> Do you have an example that compiles with ESP32? I can't get any of the examples to compile, even after changing platformio.ini.

Any progress with ESP32? Thanks

ba58smith, 2020-02-15:
@Haimana, as far as I know, no one is working on it. Sorry.

ba58smith, 2020-04-29:
See PR #115 .

ba58smith, 2020-05-25:
Closed. See PR #118.

closed 2020-05-25 by ba58smith