SignalK/SensESP #70: 'RANDOM_REG32' was not declared in this scope

LuisColling, 2019-12-04:
Follow basic readme instructions and got this error when compiling on Visual Studio Code (platformIO).

ba58smith, 2019-12-04:
Can you provide more of the error message, please? Not the entire output of the build, but the whole "paragraph" with this error in it, showing all of the files that are involved in the error. Similar to this:
```
In file included from .pio/libdeps/d1_mini/ReactESP_ID2965/src/ReactESP.cpp:3:0:
/home/butch/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FunctionalInterrupt.h:32:28: warning: 'scheduledInterrupts' defined but not used [-Wunused-variable]
 static ScheduledFunctions* scheduledInterrupts;
```

LuisColling, 2019-12-04:
Sorry Brian...I am still struggling. Decided to recreate project and now other error popped up.  

In file included from .pio\libdeps\d1_mini\SensESP\src/sensors/sensor.h:6:0,
                 from .pio\libdeps\d1_mini\SensESP\src/sensesp_app.h:7,
                 from src\main.cpp:2:
.pio\libdeps\d1_mini\SensESP\src/system/configurable.h:7:25: fatal error: ArduinoJson.h: No such file or directory

LuisColling, 2019-12-04:
The main difference from first project error to last error is the platformio.ini file.

Last error ini:

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
lib_deps = 
    https://github.com/SignalK/SensESP.git

For the RANDOM_REG32 the ini was like that:

[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
framework = arduino
lib_deps = 
    https://github.com/SignalK/SensESP.git

ba58smith, 2019-12-04:
Luis, here's my platformio.ini:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 160000000L
upload_resetmethod = nodemcu
build_flags =
   -Wl,-Teagle.flash.4m1m.ld
   -Wall
   -Wno-reorder
upload_protocol = espota   
upload_port = 192.168.1.168

upload_speed = 460800
lib_deps =
   https://github.com/SignalK/SensESP.git
   ESP Async WebServer
   ESPAsyncWiFiManager
   ReactESP@^0.2.2
   ArduinoJson@^5.0.0
   https://github.com/JoaoLopesF/RemoteDebug.git#0b5a9c1a49fd2ade0e3cadc3a3707781e819359a
   WebSockets
   DallasTemperature
   ESP8266TrueRandom
```
Try it with that, and let me know what happens.

LuisColling, 2019-12-04:
Moved to my Mac laptop with a fresh install of VSCode and platformIO. Now it is working fine. 

Most likely it was some issue with my Windows Laptop and PlatformIO installation on VSCode. Sorry about that.

Will try to reinstall the extension as well as clean the intellisense index later...

Thanks Brian!

ba58smith, 2019-12-04:
My pleasure. I'm going to close this issue. Don't hesitate to reach out again if you have trouble. I won't always be as responsive (I'm not on my computer very much - you just caught me on a good day), but I'll always try to help.

closed 2019-12-04 by ba58smith