SignalK/SensESP #71: DS18B20

kaa39518, 2020-01-04:
Please write an example of connecting a DS18B20 sensor.

ba58smith, 2020-01-04:
``
#include <Arduino.h>

//#define SERIAL_DEBUG_DISABLED

#define USE_LIB_WEBSOCKET true

#include "sensesp_app.h"
#include "sensors/onewire_temperature.h"
#include "signalk/signalk_output.h"
#include "wiring_helpers.h"

ReactESP app([] () {
  #ifndef SERIAL_DEBUG_DISABLED
  Serial.begin(115200);

  // A small arbitrary delay is required to let the
  // serial port catch up

  delay(100);
  Debug.setSerialEnabled(true);
  #endif

  sensesp_app = new SensESPApp();

  DallasTemperatureSensors* dts = new DallasTemperatureSensors(D5);

  (new OneWireTemperature(dts, "/devices/temperature"))
      -> connectTo(new Linear(1.0, 0.0, "/sensors/temperature/calibrate"))
      -> connectTo(new SKOutput<float>("propulsion.main.temperature", "/sensors/temperature"));

  (new OneWireTemperature(dts, "/devices/coolantTemperature"))
    -> connectTo(new Linear(1.0, 0.0, "/sensors/coolantTemperature/calibrate"))
    -> connectTo(new SKOutput<float>("propulsion.main.coolantTemperature", "/sensors/coolantTemperature"));

  (new OneWireTemperature(dts, "/devices/exhaustTemperature"))
    -> connectTo(new Linear(1.0, 0.0, "/sensors/exhaustTemperature/calibrate"))
    ->connectTo(new SKOutput<float>("propulsion.main.exhaustTemperature", "/sensors/exhaustTemperature"));

  sensesp_app->enable();
});
```

closed 2020-01-04 by ba58smith

kaa39518, 2020-01-05:
Earned after adding #include "transforms / linear.h" and replacing SKOutput with SKOutputNumber.
Thanks again!

ba58smith, 2020-01-05:
I believe that example was prior to a few code changes - sorry! Here it is, with those changes included:

``
#include <Arduino.h>

//#define SERIAL_DEBUG_DISABLED

#define USE_LIB_WEBSOCKET true

#include "sensesp_app.h"
#include "sensors/onewire_temperature.h"
#include "signalk/signalk_output.h"
#include "wiring_helpers.h"
#include "transforms / linear.h"

ReactESP app([] () {
  #ifndef SERIAL_DEBUG_DISABLED
  Serial.begin(115200);

  // A small arbitrary delay is required to let the
  // serial port catch up

  delay(100);
  Debug.setSerialEnabled(true);
  #endif

  sensesp_app = new SensESPApp();

  DallasTemperatureSensors* dts = new DallasTemperatureSensors(D5);

  (new OneWireTemperature(dts, "/devices/temperature"))
      -> connectTo(new Linear(1.0, 0.0, "/sensors/temperature/calibrate"))
      -> connectTo(new SKOutputNumber("propulsion.main.temperature", "/sensors/temperature"));

  (new OneWireTemperature(dts, "/devices/coolantTemperature"))
    -> connectTo(new Linear(1.0, 0.0, "/sensors/coolantTemperature/calibrate"))
    -> connectTo(new SKOutputNumber("propulsion.main.coolantTemperature", "/sensors/coolantTemperature"));

  (new OneWireTemperature(dts, "/devices/exhaustTemperature"))
    -> connectTo(new Linear(1.0, 0.0, "/sensors/exhaustTemperature/calibrate"))
    ->connectTo(new SKOutputNumber("propulsion.main.exhaustTemperature", "/sensors/exhaustTemperature"));

  sensesp_app->enable();
});
```

kaa39518, 2020-01-05:
Thank you for a great software product!
In this navigation I will test on my boat - http://24kaa.ru/

Haimana, 2020-02-11:
Hello, 
I tried the last example code, it compiles OK but in /setup url I have "Error: undefined" error and "devices" and "sensors" links are not working (links appear after pressing OK)
[image]
I'm using ESP8266 and PlatformIO v1.42.0. All the included examples are working properly.
Is it mandatory to have the DS sensors connected? Anyway, I tried both, with sensors connected and not connected. Same error.

The serial monitor output show the sensor and it's readings OK but also there are some load_configuration errors.
I also tried to erase flash and reupload. same load_configuration errors:
`(load_configuration) Not loading configuration, config_path empty or file does not exist: /system/networking
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /system/sk
␛[0m(I) (DallasTemperatureSensors) OneWire sensors found: 1
␛[0m(I) (DallasTemperatureSensors) Found OneWire sensor 28:ff:fe:1a:a2:17:04:35
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /devices/temperature
␛[0m(D) (OneWireTemperature) Registered a new OneWire sensor
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /sensors/temperature/calibrate
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.systemhz
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.freemem
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.uptime
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.ipaddr
␛[0m(I) (enable) Connecting SignalK source propulsion.main.temperature
`

After configuring WiFi:
`(load_configuration) Not loading configuration, config_path empty or file does not exist: /system/sk
␛[0m(I) (DallasTemperatureSensors) OneWire sensors found: 1
␛[0m(I) (DallasTemperatureSensors) Found OneWire sensor 28:ff:fe:1a:a2:17:04:35
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /devices/temperature
␛[0m(D) (OneWireTemperature) Registered a new OneWire sensor
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /sensors/temperature/calibrate
␛[0m(I) (enable) Connecting SignalK source sensors.onewire.systemhz
␛[0m(I) (enable) Connecting SignalK source sensors.onewire.freemem
␛[0m(I) (enable) Connecting SignalK source sensors.onewire.uptime
␛[0m(I) (enable) Connecting SignalK source sensors.onewire.ipaddr
␛[0m(I) (enable) Connecting SignalK source propulsion.main.temperature
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (enable) Subsystem: setup_discovery()
␛[0m(W) (setup_discovery) Error setting up mDNS responder
␛[0m(I) (enable) Subsystem: networking->setup()
␛[0m(I) (setup_saved_ssid) Connecting(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid)␛[0m .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_sa␛[0mved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid)
␛[0m(I)
␛[0m(I) (setup_saved_ssid) Connected to wifi, SSID: SD
␛[0m(I) (enable) Subsystem: setup_OTA()
␛[0m(I) (enable) Subsystem: http_server()
␛[0m(I) (enable) Subsystem: ws_client()
␛[0m(I) (enable) WS client enabled
␛[0m(D) (enableAll) Enabling all required sensors and transforms
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: Linear
␛[0m(D) (enableAll) Enabling sensor or transform: OneWireTemperature
␛[0m(D) (enableAll) Enabling sensor or transform: DallasTemperatureSensor
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: IPAddrDev
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: Uptime
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: FreeMem
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: SystemHz
␛[0m(I) (enable) All sensors and transforms enabled
␛[0m(D) (connect) Initiating connection
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"sensors.onewire.uptime","value":11.33},{"path":"sensor␛[0ms.onewire.freemem","value":21152},{"path":"sensors.onewire.systemhz","value":1}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"propulsion.main.temperature","value":296.2125}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"sensors.onewire.uptime","value":12.33},{"path":"sensor␛[0ms.onewire.freemem","value":21152},{"path":"sensors.onewire.systemhz","value":90039}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"propulsion.main.temperature","value":296.275}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"sensors.onewire.uptime","value":13.331},{"path":"senso␛[0mrs.onewire.systemhz","value":91071},{"path":"sensors.onewire.freemem","value":21040}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"propulsion.main.temperature","value":296.275}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"sensors.onewire.uptime","value":14.331},{"path":"senso␛[0mrs.onewire.systemhz","value":90727},{"path":"sensors.onewire.freemem","value":21040}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"onewire"},"values":[{"path":"propulsion.main.temperature","value":296.275}]}]}`

Thank you very much for your help and for your great work,
Seb

Haimana, 2020-02-11:
Hello again,
I think I solve the problem.
I add "/sk" at the end of each "SKOutputNumber" lines so my code look like this.
Not sure if this is the best solution but at least it works.
Thank you for your time

```
(new OneWireTemperature(dts, "/devices/temperature"))
-> connectTo(new Linear(1.0, 0.0, "/sensors/temperature/calibrate"))
-> connectTo(new SKOutputNumber("propulsion.main.temperature", "/sensors/temperature/sk"));

(new OneWireTemperature(dts, "/devices/coolantTemperature"))
-> connectTo(new Linear(1.0, 0.0, "/sensors/coolantTemperature/calibrate"))
-> connectTo(new SKOutputNumber("propulsion.main.coolantTemperature", "/sensors/coolantTemperature/sk"));

(new OneWireTemperature(dts, "/devices/exhaustTemperature"))
-> connectTo(new Linear(1.0, 0.0, "/sensors/exhaustTemperature/calibrate"))
->connectTo(new SKOutputNumber("propulsion.main.exhaustTemperature", "/sensors/exhaustTemperature/sk"));
```

ba58smith, 2020-02-12:
@Haimana - I'm happy you have it working, but I don't understand WHY your change made it work. In the error messages about not loading configuration, it refers to paths that I don't see in your code that you copied in: `/system/sk` and `/devices/temperature`. Also, I see that only one sensor was found, but you've got code for three sensors.

If you've been changing the paths (for the Linear() transports and the SKOutputNumber() transports), it's possible that you now have some left-over data in the config file that you might want to clean up. If you want, try this: decide on exactly what the paths are that you want (they look good now, except for the `/sk` at the ends of some of them - you shouldn't need those), make those changes, compile, and upload. Then, "log in" to the ESP8266 by entering its IP address in the browser, to bring up the Config screen. There's an option to "Reset sensor" - that will erase all configuration that you've done so far.
After you reset it, it should restart, and then you might see the `not loading configuration` messages again. If you do, go into the configuration again, and bring up the "calibrate" and look at each item. (Sorry, I'm not sure of exactly the steps, but I'm trying to get all the old configuration data erased, to be replaced by the new configuration data.)

Haimana, 2020-02-12:
Thank you for your support.
I removed "/sk" from paths and I compiled and uploaded it. After that I reset the device. Unfortunately I cannot access the calibration section because of same error.
Yes, I had one single sensor wired when I copied the monitor output  but no difference. 
Now I left code for a single sensor and also just one sensor is connected.

`load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
(I) (DallasTemperatureSensors) OneWire sensors found: 1
␛[0m(I) (DallasTemperatureSensors) Found OneWire sensor 28:ff:fe:1a:a2:17:04:35
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /devices/temperature
␛[0m(D) (OneWireTemperature) Registered a new OneWire sensor
␛[0m(I) (load_configuration) Not loading configuration, config_path empty or file does not exist: /sensors/temperature/calibrate
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.systemhz
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.freemem
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.uptime
␛[0m(I) (enable) Connecting SignalK source sensors.sensesp.ipaddr
␛[0m(I) (enable) Connecting SignalK source propulsion.main.temperature
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (enable) Subsystem: setup_discovery()
␛[0m(W) (setup_discovery) Error setting up mDNS responder
␛[0m(I) (enable) Subsystem: networking->setup()
␛[0m(I) (setup_saved_ssid) Connecting(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid)
.(setup_saved_ssid) .(setup_saved_ssid)␛[0m .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid) .(setup_saved_ssid)
␛[0m(I)
␛[0m(I) (setup_saved_ssid) Connected to wifi, SSID: SD
␛[0m(I) (enable) Subsystem: setup_OTA()
␛[0m(I) (enable) Subsystem: http_server()
␛[0m(I) (enable) Subsystem: ws_client()
␛[0m(I) (enable) WS client enabled
␛[0m(D) (enableAll) Enabling all required sensors and transforms
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: Linear
␛[0m(D) (enableAll) Enabling sensor or transform: OneWireTemperature
␛[0m(D) (enableAll) Enabling sensor or transform: DallasTemperatureSensor
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: IPAddrDev
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: Uptime
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: FreeMem
␛[0m(D) (enableAll) Enabling sensor or transform: SKOutput
␛[0m(D) (enableAll) Enabling sensor or transform: SystemHz
␛[0m(I) (enable) All sensors and transforms enabled
␛[0m(D) (connect) Initiating connection
␛[0m(I) (get_mdns_service) Found server 192.168.1.91 (port 3000)
␛[0m(D) (connect) Websocket client starting
␛[0m(D) (poll_access_request) Polling SK Server for authentication token
␛[0m(D) (poll_access_request) PENDING
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"sensesp"},"values":[{"path":"sensors.sensesp.uptime","value":7.192},{"path":"sensor␛[0ms.sensesp.freemem","value":19168},{"path":"sensors.sensesp.systemhz","value":1}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"sensesp"},"values":[{"path":"propulsion.main.temperature","value":297.3375}]}]}
␛[0m(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"sensesp"},"values":[{"path":"sensors.sensesp.uptime","value":8.192},{"path":"sensor␛[0ms.sensesp.systemhz","value":89983},{"path":"sensors.sensesp.freemem","value":19208}]}]}`

ba58smith, 2020-02-12:
Try changing `/devices/temperature` to `/sensors/temperature`. Now that I've really dug into it, I think that's the problem. (You might need to do the "Reset sensor" again, to erase all of your current configuration.

Haimana, 2020-02-14:
hello,
I tried but it is the same problem. It seems is not working if some path is contained in a different path, where act like a "subfolder", like "**/sensors/temperature**" and "**/sensors/temperature**/calibrate"
Anyway, if I add that "**/sk**" at the end, is working like a charm! :)

[image]

Thank you very much, really appreciate your work
Seb

ba58smith, 2020-03-11:
@Haimana - yes, now that I'm looking at this again, the problem was (as you said above) that you had one path for the Transport that was "/sensors/temperature/calibrate" and another that was for the Signal K Path that was just a portion of that path: "/sensors/temperature". Adding "/sk" to the end worked, but you could have added anything to the end - "/skPath" or "charlie" - it just had to be totally different from the other path.

I finally got around to adding some documentation to the GitHub Wiki page for SensESP and it talks about how to name the configuration paths.

EdKok, 2020-03-12:
@ba58smith Thanks for updating the WiKi and trying to clarify it better for us, but after reading it three times I still don't get it.
For above OneWireTemperature sensors, could you please post an example with (at least) two sensors?

ba58smith, 2020-03-12:
@EdKok - I will add this to the /examples folder later, but here is the exact code that I'm running on my boat for my four DS18B20 sensors. I just switched from SigkSens to SensESP, and it's working great. I initially had the digital pin number wrong for `DallasTemperatureSensors(D7)', and of course, it found no sensors. As soon as I fixed that, it found all four of them.

```
#include <Arduino.h>

//#define SERIAL_DEBUG_DISABLED

#define USE_LIB_WEBSOCKET true

#include "sensesp_app.h"
#include "sensors/onewire_temperature.h"
#include "signalk/signalk_output.h"
#include "wiring_helpers.h"
#include "transforms/linear.h"

ReactESP app([] () {
  #ifndef SERIAL_DEBUG_DISABLED
  Serial.begin(115200);

  // A small arbitrary delay is required to let the
  // serial port catch up

  delay(100);
  Debug.setSerialEnabled(true);
  #endif

  bool disableStandardSensors = false;

  sensesp_app = new SensESPApp(disableStandardSensors);

  DallasTemperatureSensors* dts = new DallasTemperatureSensors(D7);

  auto* pCoolantTemp = new OneWireTemperature(dts, "/coolantTemperature/oneWire");

    pCoolantTemp->connectTo(new Linear(1.0, 0.0, "/coolantTemperature/linear"))
                ->connectTo(new SKOutputNumber("propulsion.mainEngine.coolantTemperature", "/coolantTemperature/skPath"));

  auto* pExhaustTemp = new OneWireTemperature(dts, "/exhaustTemperature/oneWire");
    
    pExhaustTemp->connectTo(new Linear(1.0, 0.0, "/exhaustTemperature/linear"))
                ->connectTo(new SKOutputNumber("propulsion.mainEngine.exhaustTemperature", "/exhaustTemperature/skPath"));
  
  auto* p24VTemp = new OneWireTemperature(dts, "/24vAltTemperature/oneWire");
      
      p24VTemp->connectTo(new Linear(1.0, 0.0, "/24vAltTemperature/linear"))
              ->connectTo(new SKOutputNumber("electrical.alternators.24V.temperature", "/24vAltTemperature/skPath"));

  auto* p12VTemp = new OneWireTemperature(dts, "/12vAltTemperature/oneWire");
      
      p12VTemp->connectTo(new Linear(1.0, 0.0, "/12vAltTemperature/linear"))
              ->connectTo(new SKOutputNumber("electrical.alternators.12V.temperature", "/12vAltTemperature/skPath"));      

  sensesp_app->enable();
});
```

ba58smith, 2020-03-12:
Please let me know what part of the Wiki was confusing, so I can re-write it. In case it's the part about the config paths, which I suspect it is, you can see all of mine in the code above, and this is how they end up looking when I configure the device:
[image]

ba58smith, 2020-03-12:
The example has been uploaded to /examples: OneWire_temperatures.cpp.

Haimana, 2020-03-13:
> but you could have added anything to the end - "/skPath" or "charlie" - it just had to be totally different from the other path.
> 
> I finally got around to adding some documentation to the GitHub Wiki page for SensESP and it talks about how to name the configuration paths.

yes indeed, just to be different, sorry if I wasn't very clear
thank you for your time

BR,
Seb