SignalK/SensESP #72: setting a configurable interval for sending data to SK

Haimana, 2020-02-12:
Hello again,

There is some way to set the interval of time to send data to SK? Will be nice if is possible to set this for each used sk_path. 

Also, to disable sending SystemHZ, Uptime, Freemem and so on, there is any other way to do it?
I did it by commenting the specific lines in sensesp_app.cpp but I don;t know if this is the best way to do it.

Thank you,
Seb

ba58smith, 2020-02-12:
@Haimana, I don't think there is currently a way to set the time interval. (From what I can tell, each sensor type has its own interval.) I agree, it would be nice, as some things don't need updated multiple times per second. However, I don't think you'll run into any limitations by using the defaults.

ba58smith, 2020-02-12:
That's the only way to disable the system data (commenting out the appropriate lines).

ba58smith, 2020-03-10:
This has been done for AnalogInput with PR #76 . Once that PR has been approved, I will look at doing it for all sensors that don't have it yet.

ba58smith, 2020-03-22:
This has now been done for OneWire Temperature sensor with PR #82.

ba58smith, 2020-03-24:
All sensors for which a configurable read interval makes sense have now been done, pending review and merge of PR #76 and PR #82.

closed 2020-04-02 by ba58smith