SignalK/SensESP #73: Multiple analog inputs

Haimana, 2020-02-18:
Hello, me again! :)

Sorry for disturbing but it seems to be difficult for me to figure out how how to configure SensESP for reading multiple analog inputs/sensors. 
As far as I understand the easiest way is to useESP8266 and ADS1115 boards but I don;t have idea how to do it with SensESP.
With SigkSens it is clear but not with SensESP.
Can you provide an example?

Thank you,
Seb

ba58smith, 2020-02-18:
@Haimana (Seb) - sorry, but the code hasn't been written to use the ADS1115 board yet. I have one that I want to put in place to monitor two different alternators, but I haven't started working on it yet. Since that board is already supported in SigkSens, you can uses that now, and I'll post here when I've finished the code for SensESP. Sorry for any confusion!

Haimana, 2020-02-18:
no problem, thank you

ba58smith, 2020-03-24:
UPDATE: The code for reading from an ADS1015 or ADS1115 ADC has been written and is in PR #81 , waiting to be merged.

closed 2020-04-02 by ba58smith