SignalK/SensESP #86: "Unconfigured SensESP Device" AP name s/b unique

ba58smith, 2020-03-23:
If you happen to have two unconfigured SensESP devices, only one of them gets the AP name. It would be better if the name were unique to the device, using some part of the device's Hostname to make it unique.

This problem is made worse by the fact that some ESP's will broadcast their AP (with the name "Unconfigured SensESP Device") even after they have successfully connected to the network. (That's another issue, of course.)

ba58smith, 2020-03-25:
I've spent hours looking at this, and I can't make any progress on it. I can't sort out the confusion between the `hostname` and the `SSID`, and also, I think the SSID may be cached on the ESP or on all of my wifi-enabled devices, because even when I change the string "Unconfigured SensESP Device" to something else, it STILL broadcasts "Unconfigured SensESP Device".

mairas, 2020-04-07:
I can have a look soon(ish). Tasmota would be a good reference project for this one - the Tasmota devices advertise with an SSID like Tasmota-1234 (don't remember the exact format), where the number is unique for each sensor, probably taken from the MAC address of the wifi device.

ba58smith, 2020-04-08:
One challenge with this is that there seems to be a bug in WiFiManager in which the SSID and PW of the saved network are not erased from the flash memory on the ESP8266 when `resetSettings()` is called (which just calls `disconnect()`). This is still an open Issue with WiFiManager: https://github.com/tzapu/WiFiManager/issues/142.

ba58smith, 2020-04-13:
See PR #108.

ba58smith, 2020-04-28:
Fixed with PR #108

closed 2020-04-28 by ba58smith