SignalK/SensESP #87: Some ESP's broadcast their AP when they shouldn't

ba58smith, 2020-03-23:
This may be a firmware issue with some clones of the original Wemos D1 Mini - not sure. But some ESP8266's will activate the "Unconfigured SensESP Device" AP even after successfully connecting to the network and sending deltas to Signal K. If you connect to it, and go to 192.168.4.1, you see the config web page, the same as if you go to the IP address of the ESP. Same config web page, and both are active.

EdKok, 2020-04-17:
It is not just the case for clones.
I'm using an original Wemos D1 miniPro v1.0.0 which does this too.
That is an older type, but original: https://wiki.wemos.cc/products:retired:d1_mini_pro_v1.0.0
(as I write this the wiki is down, but it really exists)
I choose this model because of its external WiFi-antenna which is great for better connection from the engine-room.
The ESP is working perfectly with four DS18B20's attached, but vulnerably 'open' to hackers. Even worse, after connecting and going to 'system'->'networking' it shows the SSID plus password for your AP in plain text!

ba58smith, 2020-04-17:
@EdKok - I don't think it's vulnerable to hackers. The only way you can see system->networking is if you're already connected to your boat's wifi. The only time anything is open is when the Wemos is trying to connect to your wifi, and it can't, so it launches the configuration portal (with the blue buttons). And when that comes up, there is nothing in the network SSID and password fields.

ba58smith, 2020-04-17:
@EdKok - I do agree with you that it's not just clones that do it. Although I never saw it before with my genuine Wemos's, I'm seeing it now. I'm still poking around on Google for a solution, but it really is harmless, I think. Just a bit annoying.

EdKok, 2020-04-17:
> @EdKok - I don't think it's vulnerable to hackers. The only way you can see system->networking is if you're already connected to your boat's wifi. The only time anything is open is when the Wemos is [...]

I tested this today by connecting my Windows10 pc to the SSID ‘unconfigered SensESP device’ while it was serving data to signalk. After opening a browser to 192.168.4.1 it showed me all info in clear text.

ba58smith, 2020-04-17:
@EdKok - yep, you're right. I just did the same thing. Well, the good news is that someone would have to be pretty close to you to connect to the ESP's wifi. But it's still a vulnerability, and it gives me the perfect reason to open an issue on the ESPAsyncWifiManager github page.

ba58smith, 2020-09-09:
This was fixed by PR #140

closed 2020-09-09 by ba58smith