SignalK/SensESP #105: Support for DS2408, 1-Wire 8-Channel bidirectional port expander

KEGustafsson, 2020-04-12:
Would it possible to add support for DS2408 1-Wire IO expander?
[https://www.maximintegrated.com/en/products/ibutton-one-wire/memory-products/DS2408.html](url)

This would add more IO's to be monitored/controlled and possibility to place sensors close to sources more far away from SensESP (1-Wire interface).

ba58smith, 2020-04-12:
Sorry, I read the description of the device, and I don't understand why you would want to use it, as it all appears to require a lot of wires to be run to all the "Devices". The point of SensESP is to use inexpensive wifi devices (Wemos D1 Mini ESP8266) placed wherever you need to collect data and send data to SK via wifi. Adding code to allow you to read a whole bunch of wired devices seems inconsistent with the objective of SensESP. Am I missing something?

KEGustafsson, 2020-04-12:
Wemos current consumption is not small, 120-170mA depending on used protocol (b,g,n). If I would add Wemos to every point I would like to measure, then I'll easily end up to many Wemos devices and huge current consumption. By supporting local sensoring with one Wemos and 1-Wire expansions, we can reduce numbers of Wemos devices and reduce current consumption. Same applies to the other issue that I generated.

This is how I'm building up my sensor network with my current microcontroller setup (PIC based). I was thinking to migrate to SensESP if power consumption would be reasonable. I have sensor which are powered 24/7/365 and some which will be powered when I'm using my boat.

ba58smith, 2020-04-12:
No, SensESP doesn't put the ESP to sleep or turn wifi on/off. Given that most boats have more than enough of a "house bank" of 12V batteries to power an army of ESP's, power saving has not been considered. 

If you want to try to write the code for the DS28E17 and submit a Pull Request, look at the files https://github.com/SignalK/SensESP/blob/master/src/sensors/onewire_temperature.h and https://github.com/SignalK/SensESP/blob/master/src/sensors/onewire_temperature.cpp. You'll see how multiple 1-Wire devices are handled currently, and you may be able to copy-and-modify that code to work with the DS28E17.

closed 2020-07-29 by KEGustafsson