SignalK/SensESP #106: Support for DS28E17, 1-Wire-to-I2C Master Bridge

KEGustafsson, 2020-04-12:
Would it be possible to add support for DS28E17 1-Wire-to-I2C Master Bridge?
[https://www.maximintegrated.com/en/products/ibutton-one-wire/ibutton/DS28E17.html](url)

This would add I2C control over 1-Wire interface and freedom to place I2C devices more far away from SensESP.

E.g. Use case: Control of one/many ADS1x15 4-channel ADC devices over 1-Wire interface?

ba58smith, 2020-04-12:
Similar to my comment on your other Issue #105 - this would seem to be a way to use really long wires to get data from sensors that are far away from your data gathering. The point of using wifi is to avoid the wires altogether. Put a bunch of inexpensive ESP's all over your boat with a bunch of sensors attached to each one (sensing data that is near each one), and then send the data to SK via wifi. Am I missing something?

KEGustafsson, 2020-04-12:
I added comments to #105, which applies also here.

ba58smith, 2020-04-12:
I did, too.

closed 2020-07-29 by KEGustafsson