SignalK/SensESP #111: Add support for INA226 DC current sensor

ba58smith, 2020-04-16:
Requested by someone on Slack.

ba58smith, 2020-04-28:
Having issues with the INA226 library. Don't merge this until that is resolved.

ba58smith, 2021-01-07:
UPDATE: I've created a new Sensor (INA2xx) that will work with ALL of the INA2xx sensors, as well as the INA3221 3-channel sensor. It's being created as the first "external SensESP library". When it's finished, it will close this Issue. See https://github.com/SensESP/INA2xx.

closed 2021-12-24 by ba58smith

AAlbero, 2022-09-25:
@ba58smith I was not able to find this. The link you posted is 404. Can you help? Anything I do to add INA226 in my SensESP projects place my SH-ESP32 in a reboot loop. Thanks.

ba58smith, 2022-09-25:
Sorry, I can't find it either. I'm guessing it was moved or made obsolete when the whole SensESP project was reorganized several months ago. Maybe @mairas can remember what happened to it.