SignalK/SensESP #129: ESP32 only: Guru Meditation Error: Core  1 panic'ed (LoadProhibited).

ba58smith, 2020-05-25:
This happens only when using ESP32, and only when the SK Server isn't responding to the mDNS service request, or when the ESP doesn't yet have security authorization from the SK Server.

Stopping/starting the SK Server solves the first problem.
Authorizing the security request on the SK Server solves the second problem.

But it would be good to have these exceptions handled properly. Now, the ESP restarts, and just keeps coming back to the same spot, where it errors out again, and again, and again, until you stop/start the Server, or authorize the security request.

JohnySeven, 2020-08-05:
@ba58smith can you assign this to me?

tkurki, 2020-08-05:
@JohnySeven just curious - why? 

A practise that I've used elsewhere is that you open a [WIP] marked pull request that says `Fixes #xxx` as soon as you have something meaningful. Github takes care of linking the issue and the pr. This shows that (a) there's somebody working on this (b) others can have a look sooner rather than later ( for example to avoid "oh damn, my plan was to remove that structure, now you've built more on it") (c) no need to deassign (if work stalls it is pretty clear, no more commits to the PR - I have seen plenty of issues that are left assigned but in reality are abandoned, causing _hey are you actually working on this_ communication).

JohnySeven, 2020-08-05:
@tkurki I'll look into it. :-) But I can do it you way no problem and make PR first. Thanks

bwssytems, 2020-08-16:
If you need any help on this issue, I'll try to take a look...

JohnySeven, 2020-08-16:
I'm about to start looking at the issue, but thanks for offer . Any help would be appreciated 🙂

bwssytems, 2020-08-17:
An observation is that when mDNS is not detected the ReactESP loop for running ws_client::connect kills it as it returns without connecting. Also, the same thing happens when it tries to get authorization it just dies with the exception. If we cannot get a good connection or the authorization, it just needs to loop until it does and not die. Another items is if there is no mDNS, we can just set the use_mDNS flag to false and it will be able to go to the

bwssytems, 2020-08-17:
Well figured it out. I will fork the master and submit a pull request. It is in the ws_client::loop function that does not check if it is connected first....

bwssytems, 2020-08-17:
Added pull request #146

JohnySeven, 2020-08-17:
@bwssytems I came to same conclusion, but you were faster with fix. Nice 👍

bwssytems, 2020-08-17:
Not quite, there is another state issue.... working on it now

bwssytems, 2020-08-18:
I updated my version and working much better...

ba58smith, 2020-09-13:
Closed with PR #146

closed 2020-09-13 by ba58smith