SignalK/SensESP #130: ESP32: Error: The program size is greater than maximum allowed

free-x, 2020-05-26:
Hi,
I am trying to get my ESP32 running with BME280. Unfortunately firmware gets too big
Compiler log can you find in attachment
[SensESP.txt](https://github.com/SignalK/SensESP/files/4684427/SensESP.txt)

Regards
free-x

andyrbarrow, 2020-05-26:
Same here. Love to know the solution.

ba58smith, 2020-05-26:
See the README document, which refers to this platformio.ini file, which addresses the problem:
https://github.com/SignalK/SensESP/blob/master/platformio.ini
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
; Un-comment the following if you get an error about
; insufficient memory after compiling and linking:
; board_build.partitions = no_ota.csv

; Un-comment the following if you want more 
; details about runtime errors:
; monitor_filters = esp32_exception_decoder
```

ba58smith, 2020-05-26:
@free-x and @andyrbarrow - please post here the ESP32 board you're using, the line in platformio.ini that identifies it (in my case it's `[env:esp32dev]`), and whether or not this fixes the problem. If it does, I'll update the README to tell how to do this.

andyrbarrow, 2020-05-27:
I changed board to esp32dev and I have the same error.
> Executing task in folder SensESPTest: platformio run <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SensESP> 0.4.3 #de1c0c8
|   |-- <ReactESP> 0.2.2
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESP Async WebServer> 1.2.3 #f13685e
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 5.13.4
|   |-- <ESPAsyncWiFiManager> 0.22
|   |   |-- <DNSServer> 1.1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <ESP Async WebServer> 1.2.3 #f13685e
|   |   |   |-- <AsyncTCP> 1.1.1
|   |   |   |-- <FS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <ArduinoJson> 5.13.4
|   |   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|   |-- <WebSockets> 2.2.1
|   |   |-- <WiFi> 1.0
|   |   |-- <SPI> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <OneWire> 2.3.5
|   |-- <DallasTemperature> 3.8.1
|   |   |-- <OneWire> 2.3.5
|   |-- <ESPTrueRandom> 1.1 #ed198f4
|   |-- <Wire> 1.0.1
|   |-- <Adafruit ADS1X15> 1.1.0
|   |   |-- <Wire> 1.0.1
|   |-- <Adafruit BME280 Library> 2.0.2
|   |   |-- <Adafruit Unified Sensor> 1.1.2
|   |   |   |-- <Adafruit ADXL343> 1.3.0
|   |   |   |   |-- <Adafruit BusIO> 1.2.6
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |-- <Adafruit BMP280 Library> 2.0.1
|   |   |-- <Adafruit Unified Sensor> 1.1.2
|   |   |   |-- <Adafruit ADXL343> 1.3.0
|   |   |   |   |-- <Adafruit BusIO> 1.2.6
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <Wire> 1.0.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Adafruit SHT31 Library> 1.2.0
|   |   |-- <Wire> 1.0.1
|   |-- <Adafruit INA219> 1.0.9
|   |   |-- <Adafruit NeoPixel> 1.4.0
|   |   |-- <Adafruit GFX Library> 1.8.3
|   |   |   |-- <Adafruit BusIO> 1.2.6
|   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |   |-- <Adafruit SSD1306> 2.2.1
|   |   |   |-- <Adafruit GFX Library> 1.8.3
|   |   |   |   |-- <Adafruit BusIO> 1.2.6
|   |   |   |   |   |-- <SPI> 1.0
|   |   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |   |-- <Adafruit BusIO> 1.2.6
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |   |-- <Wire> 1.0.1
|   |-- <RemoteDebug> 2.1.2 #0b5a9c1
|   |   |-- <WiFi> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|-- <RemoteDebug> 2.1.2 #0b5a9c1
|   |-- <WiFi> 1.0
Building in release mode
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1339349 bytes) is greater than maximum allowed (1310720 bytes)
RAM:   [=         ]  13.7% (used 44792 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  102.2% (used 1339349 bytes from 1310720 bytes)
============================================== [FAILED] Took 3.5

free-x, 2020-05-27:
i have tried following boards
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = https://github.com/SignalK/SensESP.git
monitor_speed = 115200
upload_speed = 115200

[env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
framework = arduino
lib_deps = https://github.com/SignalK/SensESP.git
monitor_speed = 115200
upload_speed = 115200
```
Both envs failed

ba58smith, 2020-05-27:
I should have been more specific. What I wanted you to do was to add this line to the `[env:]` section of your platformio.ini: `board_build.partitions = no_ota.csv`. (That's what you'll see as the cure for the insufficient memory error in this file: https://github.com/SignalK/SensESP/blob/master/platformio.ini).

I didn't want you to change the board to esp32dev - I simply asked what board you were using. But now, why don't you tell me exactly what board you're using, by providing a url to where you bought it.(In the meantime, try `[env: esp32dev]` with the above line added to that section.)

andyrbarrow, 2020-05-27:
I added that line to esp32dev and it built successfully. 

I'm going to try other devices now. My goal is to get this working with M5StickC. There are some GPIO mapping issues that need to be resolved, but if I can get it working, I think it will be an excellent platform for SensESP.

free-x, 2020-05-27:
@ba58smith 
I interpreted your example
```
; board_build.partitions = no_ota.csv
```
for me is this only comment. It's not active option

I'm using AZ-Delivery ( https://www.az-delivery.de/collections/more-products-1/products/esp32-developmentboard)

free-x, 2020-05-27:
success. with "**board_build.partitions = no_ota.csv**"

Another thing: some ESP32 boards doesn't provide BUILTIN_LED
My workaround in src/system/led_blinker.cpp
```
#ifndef LED_BUILTIN
#define LED_BUILTIN 1
#endif
```

ba58smith, 2020-05-27:
> I added that line to esp32dev and it built successfully.
> 
> I'm going to try other devices now. My goal is to get this working with M5StickC. There are some GPIO mapping issues that need to be resolved, but if I can get it working, I think it will be an excellent platform for SensESP.

I saw an M5Stick a few days ago, but it wasn't clear to me how I would use it. Two of my three ESP8266's have multiple sensors connected, each with a breakout board or a resistor. How would that work with the M5Stick?

ba58smith, 2020-05-27:
> @ba58smith
> I interpreted your example
> ```
> ; board_build.partitions = no_ota.csv
> ```
> for me is this only comment. It's not active option

No, it's not active with the semicolon. But the two lines above tell you to "uncomment the line" (remove the semicolon) if you get an "insufficient memory " error. No matter - it's working now.

ba58smith, 2020-05-27:
> success. with "**board_build.partitions = no_ota.csv**"
> 
> Another thing: some ESP32 boards doesn't provide LED_BUILTIN
> My workaround in src/system/led_blinker.cpp
> 
> ```
> #ifndef LED_BUILTIN
> #define LED_BUILTIN 1
> #endif
> ```
> 
> This works with AZ-delivery ESP32

Right. That's why the comments are there in led_blinker.cpp. Did defining your LED_BUILTIN as 1 make an onboarding LED flash, or did it just solve the compiler error?

free-x, 2020-05-27:
> Did defining your LED_BUILTIN as 1 make an onboarding LED flash, or did it just solve the compiler error?
according to documentation for my kit

ba58smith, 2020-05-29:
I updated the README to (hopefully) make it clear how to address this issue if necessary for your particular ESP32 board.

closed 2020-05-29 by ba58smith