SignalK/SensESP #136: Support ESP01 1Mb flash, options to trim memory usage

barrymac, 2020-07-04:
I'm trying to flash an ESP8266 ESP01 unit with 1M flash, and get the following error:

```
Checking size .pio/build/esp01_1m/firmware.elf
Error: The program size (786016 bytes) is greater than maximum allowed (761840 bytes)
```

I have uncommented the line 

`board_build.partitions = no_ota.csv`

in the platformio.ini file

ba58smith, 2020-07-06:
That line in platformio.ini is in the section you would be using if you are using an ESP32 microcontroller. You say you're using an ESP8266 microcontroller. Can you copy your platformio.ini file here, please?

barrymac, 2020-07-09:
Thank you for the response. 

`[env:esp01_1m]
platform = espressif8266
board = esp01_1m
monitor_speed = 115200
framework = arduino
lib_deps =
    https://github.com/SignalK/SensESP.git

board_build.partitions = no_ota.csv
`

ba58smith, 2020-07-10:
@barrymac - I can't find anything that indicates no_ota.csv works with ESP8266 - it seems to be only an ESP32 thing.

It may simply be that SensESP has now exceeded the size limitations of the 1 meg ESP8266 boards. One of my SensESP programs (4 1-wire temp sensors and 1 digital pin input) is 790,744 bytes, and all of mine are about the same size. That's probably too big for an MCU with 1 meg of RAM. Try it with a 4 meg version of the chip, such as the D1 Mini (NOT the D1 Mine Lite!).

barrymac, 2020-07-10:
I am looking for an educated way to trim down some features that I won't need in this unit, and save some space in case it is too big for 1M. My first use case is a simple temperature monitor, and so it won't need support for all the other types of sensors. I'm not sure where to start though, as I'm totally new to the field of MCU development. I've forked the repo and will attempt to trim it down and use that as a dependency instead of the main repo.

ba58smith, 2020-07-10:
@barrymac - I'm pretty sure that PlatformIO includes on the libraries it needs to build your particular main.cpp, so I'm not sure how much trimming you can do. Is it really worth the effort? You can buy the 4 meg versions for under US$5 each. (https://smile.amazon.com/IZOKEE-NodeMcu-Internet-Development-Compatible/dp/B076F52NQD)

If you're intent on trying this, you might look at the OTA (over the air updating) code - I have no idea how much space it will save in the end, but I'm sure you can't use it with a 1 meg MCU, so you might as well remove it.

JohnySeven, 2020-08-05:
Have you tried playing with these parameters: https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size ?

mairas, 2020-08-07:
Assuming this won't come too late: The debugging library is kinda big and
if you don't use the remote debugging features, it's quite unnecessary. I
suppose you could squeeze the code size under the limit with that. You
could reimplement the debug print functions (debugI etc) with simple
functions or macros.

ibravo, 2020-09-13:
When looking at the compilation code, it looks like the SensESP is compiling for all sensors, whether they will be used or not.
I'm just getting started with this project, and following the wiki, I'm implementing an Analog sensor, but the code seems to compile them all:

Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/bme280.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/bmp280.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/digital_input.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/digital_output.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/gps.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/ina219.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/max31856TC_input.cpp.o
Compiling .pio/build/esp01_1m/liba42/SensESP/sensors/onewire_temperature.cpp.o

Is there a way to turn on/off which sensors to include in the final bin file?

ba58smith, 2020-09-14:
It compiles them all, but doesn't include them all in the final binary - only the ones that are needed by main.cpp.

ba58smith, 2020-12-10:
This has become very stale, and isn't a bug, nor a feature that seems to have any real support, so I'm going to close it.

closed 2020-12-10 by ba58smith