SignalK/SensESP #144: Can't set hostname in main.cpp w/o also setting ssid and pw

ba58smith, 2020-08-16:
The new SensespAppOptions feature that allows various options to be set in main.cpp has a problem if you try to set the SSID and password in main.cpp, but DON'T set the hostname in main.cpp. This is most likely due to the fact that WifiManager saves the hostname as a custom parameter, so when you don't use WifiManager to set SSID and pw, you don't get access to hostname, because WifiManager isn't involved.
I think an acceptable solution would be to bundle these three options: either set all three in main.cpp, or set none of them in main.cpp. But that's up for discussion.

mairas, 2020-08-24:
Could you verify that this is still the case after my related modifications?

ba58smith, 2020-08-31:
@mairas, it actually seems much worse now. With this in main.cpp:
```
  // Create the global SensESPApp() object.
  sensesp_app = builder.set_standard_sensors(UPTIME)
              //->set_hostname("d1minitest")
              //->set_sk_server("10.10.10.1", 3000)
              ->set_wifi("contemporary", "heatheranderic")
              ->get_app();
```
the program uploads and runs, and as soon as it gets to `setup_discovery( )`, it crashes hard. Below is everything from "Linking" to the crash.

Linking .pio/build/d1_mini/firmware.elf
Building .pio/build/d1_mini/firmware.bin
Retrieving maximum program size .pio/build/d1_mini/firmware.elf
Checking size .pio/build/d1_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.4% (used 41324 bytes from 81920 bytes)
Flash: [========  ]  76.2% (used 795856 bytes from 1044464 bytes)
Creating BIN file ".pio/build/d1_mini/firmware.bin" using "/home/butch/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and ".pio/build/d1_mini/firmware.elf"
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...

Warning! Your `/etc/udev/rules.d/99-platformio-udev.rules` are outdated. Please update or reinstall them.
 Mode details: https://docs.platformio.org/en/latest/faq.html#platformio-udev-rules

Auto-detected: /dev/ttyUSB0
Uploading .pio/build/d1_mini/firmware.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: bc:dd:c2:9e:2a:91
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 800016 bytes to 456334...
Wrote 800016 bytes (456334 compressed) at 0x00000000 in 10.3 seconds (effective 621.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================================================= [SUCCESS] Took 21.81 seconds =============================================================================
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
(I) (operator()) Serial debug enabled
␛[0m(I) (Networking) Using hard-coded SSID contemporary and password
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/sk
␛[0m(I) (DallasTemperatureSensors) Found OneWire sensor 28:d6:35:79:97:17:03:37
␛[0m(I) (DallasTemperatureSensors) Found OneWire sensor 28:21:bb:79:97:16:03:6d
␛[0m(I) (DallasTemperatureSensors) Found OneWire sensor 28:e7:cd:79:97:16:03:b1
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /coolantTemperature/oneWire
␛[0m(D) (OneWireTemperature) Registered a new OneWire sensor
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /coolantTemperature/linear
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /exhaustTemperature/oneWire
␛[0m(D) (OneWireTemperature) Registered a new OneWire sensor
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /exhaustTemperature/linear
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /24vAltTemperature/oneWire
␛[0m(D) (OneWireTemperature) Registered a new OneWire sensor
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /24vAltTemperature/linear
␛[0m(I) (enable) Connecting SignalK source .uptime
␛[0m(I) (enable) Connecting SignalK source propulsion.mainEngine.coolantTemperature
␛[0m(I) (enable) Connecting SignalK source propulsion.mainEngine.exhaustTemperature
␛[0m(I) (enable) Connecting SignalK source electrical.alternators.24V.temperature
␛[0m(I) (enable) Connecting SignalK source electrical.erBilgePump.state
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (enable) Subsystem: setup_discovery()
␛[0m(I) (setup_discovery) mDNS responder started at 
␛[0m
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc10 end: 3fffffc0 offset: 0190
3ffffda0:  3ffffe20 00000050 20746120 0000000a  
3ffffdb0:  0000000b 00000001 40221b08 3fff1f20  
3ffffdc0:  3ffffe20 3fff50b4 3fff4c0c 4021fc33  
3ffffdd0:  00000050 0000000f 3ffffe2c 4020f72b  
3ffffde0:  00000003 000000db 3fff1f20 3fff50b4  
3ffffdf0:  3ffffe20 00000050 3fff1f20 4021fcd6  
3ffffe00:  3fff34b0 3ffffe20 3ffea040 3fff4d78

ba58smith, 2020-08-31:
It does the same thing with this in main.cpp (not setting anything other than standard sensors). (I did "Erase flash" between these tests.)
```
 // Create the global SensESPApp() object.
  sensesp_app = builder.set_standard_sensors(UPTIME)
              //->set_hostname("d1minitest")
              //->set_sk_server("10.10.10.1", 3000)
              //->set_wifi("contemporary", "heatheranderic")
              ->get_app();
```

ba58smith, 2020-08-31:
But with this in main.cpp (setting standard sensors and hostname), everything works as it should: the "Configure diminitest" AP appears, I connnect to it, go to 192.168.4.1, configure the wifi, it restarts, connects to wifi, then gets to "All sensors and transforms enabled", then "Initiating connection", then "No SignalK server found in network when using mDNS service!", and just repeats those last two lines every 10 seconds.
```
  // Create the global SensESPApp() object.
  sensesp_app = builder.set_standard_sensors(UPTIME)
              ->set_hostname("d1minitest")
              //->set_sk_server("10.10.10.1", 3000)
              //->set_wifi("contemporary", "heatheranderic")
              ->get_app();
```

ba58smith, 2020-08-31:
This works as expected:
```
  // Create the global SensESPApp() object.
  sensesp_app = builder.set_hostname("d1minitest")
              //->set_standard_sensors(UPTIME)
              //->set_sk_server("10.10.10.1", 3000)
              //->set_wifi("contemporary", "heatheranderic")
              ->get_app();
```

But this goes into the crash loop:
```
  // Create the global SensESPApp() object.
  sensesp_app = builder.get_app(); 
              //set_hostname("d1minitest")
              //->set_standard_sensors(UPTIME)
              //->set_sk_server("10.10.10.1", 3000)
              //->set_wifi("contemporary", "heatheranderic")
              // ->get_app();
```

ba58smith, 2020-08-31:
Conclusion: if you don't set the hostname in main.cpp, it crashes.

All of this was tested with a Wemos D1 Mini.

ba58smith, 2020-08-31:
This doesn't crash, but since I'm not connected to a SK server (I'm away from the boat), it just keeps trying "Initiating connection" every 10 seconds, as it should.
```
  // Create the global SensESPApp() object.
  sensesp_app = builder.set_hostname("d1minitest")
              ->set_standard_sensors(UPTIME)
              ->set_sk_server("10.10.10.1", 3000)
              ->set_wifi("contemporary", "heatheranderic")
              ->get_app();
```

ba58smith, 2020-09-04:
Finally had a chance to look into this, and I find this in `sensesp_app.cpp`:
```
debugI("Enabling subsystems");

  // FIXME: Setting up mDNS discovery before networking can't work!

  debugI("Subsystem: setup_discovery()");
  setup_discovery(networking->get_hostname()->get().c_str());

  debugI("Subsystem: networking->setup()");
  networking->setup([this](bool connected) {
...

```
The "FIXME" comment seems like it could be pointing to the problem.

ba58smith, 2020-09-04:
The crash is solved by providing a default hostname in sensesp_app_builder.h:
```
class SensESPAppBuilder {
 private:
  String hostname = "SensESP";
```
Apparently, MDNS.begin(hostname) doesn't like it when "hostname" is an empty string.

ba58smith, 2020-09-04:
The bug still exists: if you use `SensESPAppBuilder::set_wifi()` in main.cpp, that makes the hostname read-only. But I think I fixed it. Will post a PR later.

ba58smith, 2020-09-09:
Fixed by PR #157. In the process, the crash that was identified above was fixed by PR #155.

closed 2020-09-09 by ba58smith