SignalK/SensESP #151: BME280, SensESP fails to start if sensor not present

KEGustafsson, 2020-08-26:
If BME280 device is not present then SensESP startup fails and device is halted.
␛[0m(I) (check_status) Could not find a valid BME280 sensor. Check wiring, address, and sensor ID. ␛[0m(I) (check_status) SensorID is: 0x255 ␛[0m(I) (check_status) 0xFF: is a BMP180 or BMP085, or a bad address ␛[0m(I) (check_status) 0x56-0x58 is a BMP280 ␛[0m(I) (check_status) 0x60 is a BME280 ␛[0m(I) (check_status) 0x61 is a BME680 ␛[0m
Log reporting stops here.

Program stops here.
https://github.com/SignalK/SensESP/blob/master/src/sensors/bme280.cpp#L27

Would be much better to catch the issue internally and just disable reading the values for the sensor and allow the rest of the program to continue.

BjarneBitscrambler, 2020-08-29:
This behaviour will also happen using the orientation sensor I've been working on ( #145 ).  I'll have a look at my code and see about fixing it - I agree that a system that potentially has dozens of sensors should degrade gracefully if any one of the sensors fails.

ba58smith, 2020-09-09:
@BjarneBitscrambler - I've fixed this behavior for the BME280 sensor with PR #158. I'll leave it up to you to create an issue for your magnetometer sensor, or just fix it in another PR.

closed 2020-09-09 by ba58smith