SignalK/SensESP #153: Need a better approach to maintain the platformio.ini file

mairas, 2020-09-07:
`platformio.ini` defines the build environment settings for the project. Whenever anyone wants to develop SensESP itself, they usually need to modify the file. This is not ideal because maintaining the local modications of a file tracked by Git always requires some extra effort. Ideally, we should have a way to keep a default `platformio.ini` file in the repository while allowing developers to have their own local versions for their own use.

Proposal: We should rename `platformio.ini` to `platformio.ini.default`. Anyone who wants to poke at SensESP then needs to make their own copy of platformio.ini but once done, they wouldn't need to worry about changing the global defaults. Downside of the approach would be the new setup chore before a person unfamiliar to the project can get up to speed.

ba58smith, 2020-09-07:
I really like having the program as close as possible to usable by a new user, after cloning it from GitHub, so having a platformio.ini that already knows how to work with some of the most common ESP's (Wemos D1 Mini, etc.) is important. Then the README can provide very basic instructions for specifying which of those a new user is using, and it should just work. I've see a few PR's that have included such a platformio.ini, but we haven't merged any of them yet. They have a section for all of the settings necessary regardless of the board, then a section for each board's settings, and all it takes is one change to make it work with any of the boards. (https://github.com/SignalK/SensESP/blob/149aeb921dccdac1f7e37143099f06d614d0ddd8/platformio.ini)
I'm all in favor of adapting something like this, and then making it "off limits" for developers unless it's done as a separate PR specifically for adding support for another board.

KEGustafsson, 2020-09-07:
If you add platformio.ini to .gitignore -file, then PRs would not contain it. Modifications to platformio.ini could be done from webui. With git cloning, you'll get ini-file.

It would be nice to have as many platforms included to ini-file and default could be Wemos D1 mini as it is now.

mairas, 2020-09-08:
@KEGustafsson that would be an ideal solution but I don't think you can ignore files already tracked by git. And yes, quick testing unfortunately verified that.

OTOH, if @ba58smith thinks this really isn't an issue, maybe we could just live with it?

ba58smith, 2020-09-08:
@mairas - Currently, you and I are the only ones merging PR's. If we just don't allow platformio.ini to be included in any PR, except a PR specifically done to update the default platformio.ini - would that work, at least for now?

mairas, 2020-09-09:
> @mairas - Currently, you and I are the only ones merging PR's. If we just don't allow platformio.ini to be included in any PR, except a PR specifically done to update the default platformio.ini - would that work, at least for now?

I'm sure it would. But to be precise, I only want to avoid accidental updates. Improvements and other conscious modifications are fair game, as long as their impact is considered!

On the other side of the scale is your point that there should be a default platformio.ini file in place. Given that it's likely that more people download and build the source than make PRs, ease of getting started should be our priority. I'll just close this issue!

closed 2020-09-09 by mairas