SignalK/SensESP #156: Options in platformio.ini have been deprecated

ba58smith, 2020-09-08:
Near the end of a build, I get: `Warning! '-Wl,-T' option for specifying linker scripts is deprecated. Please use 'board_build.ldscript' option in your 'platformio.ini' file.`

I've looked a bit on the PIO website for something that explains how to convert from the existing build flag `-Wl,-Teagle.flash.4m1m.ld` to the equivalent with `board_build`, but I can't figure it out.

I'm afraid if we don't address this, we're going to get a breaking update to PIO one of these days.

tedenda, 2020-10-07:
According to this [documentation](https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size
) it should be done like this:

`board_build.ldscript = eagle.flash.4m.ld`

I tried that now:
```
[espressif8266_base]
;this section has config items common to all ESP8266 boards
platform = espressif8266
board_build.ldscript = eagle.flash.4m.ld
build_flags =
   -Wall
   -Wno-reorder
;   -Wl,-Teagle.flash.4m1m.ld
```

I don't get any error at compile, firmware uploads to the board.
But the filesystem seem to be faulty:

```
~ld
(I) (operator()) Serial debug enabled
␛[0m(E) (SensESPApp) FATAL: Filesystem initialization failed.
␛[0m
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
```

Just to be sure I erased flash memory and tried again, same thing.

ba58smith, 2020-11-03:
I think it should be `board_build.ldscript = eagle.flash.4m1m.ld`, to match what's being replaced: `-Teagle.flash.4m1m.ld`. I'll give that a try.

tedenda, 2020-11-03:
@ba58smith I missed that back then, good that you found it. Probably works now as it should...

ba58smith, 2020-11-03:
It seems to work. I'm creating a PR now.

tedenda, 2020-11-04:
This issue is now sorted out and can be closed (PR #228 is merged).

ba58smith, 2020-11-04:
Closed with PR #228

closed 2020-11-04 by ba58smith