SignalK/SensESP #173: Allow device to fill in requested permissions on token request to server

JohnySeven, 2020-09-26:
Hello,

it's just small think but it will allow devices to define what SK permissions they need in order to work, some of them needs just "readonly", mostly they need "readwrite" or "admin".

My sugestion is to change the token request to provide "permissions" in JSON token request and add new method in SensESP builder to configure this. Default permission will be "readwrite" as most devices needs to pull the data in.

This scenario is supported by SK server here:
`https://github.com/SignalK/signalk-server/blob/1c80214bd86272c04c47b8cf8be363e617d202a3/src/tokensecurity.js#L1035`

Just let me know what you think about this.

I'm willing to make PR.

Thanks,
Jan

ba58smith, 2020-09-29:
I like this - it will mean one less step in getting a new ESP sending data to Signal K.

mairas, 2020-09-29:
Oh, I didn't know that was supported! That's annoyed me forever. I'd love it too.

ba58smith, 2020-12-10:
Closed with PR #201.

closed 2020-12-10 by ba58smith