SignalK/SensESP #178: ESP32 Analog input noisy and inacurate resuts

GaryWSmith, 2020-09-30:
Poor support for ESP32 Analog input.

Line 14 in analogvoltage.cpp assumes an ADC resolution of 1024 (as do comments in the temperature_sender example) whereas the ESP32 ADC's have a resolution of 4095. The ESP 8266 ADC does have a resolution of 1024.

But the problem with ESP32 support is more fundamental. The code does not seem to make use of the factory calibration data and results are both noisy and inaccurate.

See this link for code with stable and accurate ADC results
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html

ba58smith, 2020-10-01:
In addition to the link above that describes the technical details of the ESP32's ADC, this link provides a good example of code that would be used in the ESP32-specific code for reading the ADC and converting it to an accurate voltage reading: https://madhephaestus.github.io/ESP32AnalogRead/ESP32AnalogRead_8cpp_source.html#l00022

ba58smith, 2020-10-01:
PR #181 deals with the 1024 vs. 4096 resolution of the two chips. All the other code referred to above still needs to be added.

GaryWSmith, 2020-10-04:
The correct way to handle the ESP32 analog requires a separate analog_input sensor class.

https://github.com/SignalK/SensESP/pull/183

ba58smith, 2020-10-11:
See PR #186. Closing this issue.

closed 2020-10-11 by ba58smith