SignalK/SensESP #179: Various compiler warnings that should be dealt with

ba58smith, 2020-10-01:
1. Library Manager: Warning! More than one package has been found by OneWire requirements:
- It lists a bunch, but defaults to the first one, by "paulstoffregen". It works, and is the one we've always used, so we should specify this one: `paulstoffregen/OneWire` will always use the latest version of that library, which is what we should probably use.

2. bool HTTPClient::begin(String) is deprecated.
- This refers to the several lines in `ws_client.cpp` that call the above, with `http.begin(url)`. We need to figure out what the current signature is for `begin()` and use it instead.

3. 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.
- In `sensesp_app.cpp`, `configurable.cpp`, `spiffs_storage.cpp`.

mairas, 2020-10-02:
SPIFFS deprecation is going to be interesting. I had a quick look at LittleFS and while it seems technologically superior, every file reserves a minimum of 4 KiB of "disk space". Since we in a typical configuration only have 64 KiB to begin with, it means there's only room for 16 files. At the moment we save the sensor/transform config filenames equal to their config paths, which in SPIFFS are single files but in LittleFS would be several (I assume one directory would be one file). So, it's possible we'd easily run out of disk space unless the config saving approach gets changed. Not a showstopper but something to keep in mind.

ba58smith, 2020-10-03:
@mairas - Would we change the config saving so that there would be only one file, which would be a single JSON file, with all the JSON representing the individual configuration items (each sensor and transform, the wifi credentials, hostname, etc.) being added together to make one big nested array, or an array of nested arrays?

ba58smith, 2020-10-03:
Looking at https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#filesystem, I wonder if we really do need to worry about using too much memory. Maybe with an ESP8266 with 1 meg, but we've set the minimum to be the 4 meg version, so if it doesn't work with the 1 meg version, I wouldn't feel bad about that. And with the 4 meg version, the minimum file system size is 1 meg. That's 256 files if each one is 4K.

mairas, 2020-10-03:
A single config file might be awkward to deal with. You'd need to store all of that in memory and parsing would be more complex. But you're right; I was calculating based on 64k filesystem size and in practice it's always larger than that. On ESP8266 our default is 1M and while the currently used min_spiffs partition layout on ESP32 provides only 180K for the filesystem, it would be quite simple to customize that.

ba58smith, 2020-10-03:
The instructions for migrating to littleFS look very straightforward. I'll give it a go - and hopefully will have more success than my ArduinoJson 6 migration attempt! :-)

ba58smith, 2020-10-03:
RE: HTTPClient::begin(String) - see https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino for an example of how to do it now.

ba58smith, 2020-10-06:
#1 above is addressed by PR #188 .

ba58smith, 2020-10-06:
#2 above is addressed by PR #191 .

tedenda, 2020-10-11:
> #1 above is addressed by PR #188 .

By using `#1` you reference another issue.
Maybe not a big deal, but anyway...

ba58smith, 2020-10-11:
> > #1 above is addressed by PR #188 .
> 
> By using `#1` you reference another issue.
> Maybe not a big deal, but anyway...
> Maybe a use a task list in the first post instead in future?

Ah yes... Fixed.

ba58smith, 2020-10-12:
While the ESP8266 has deprecated SPIFFS, ESP32 has not yet even written an interface for LittleFS. So at this point, it would have to be a case of using SPIFFS for ESP32 while switching ESP8266 to LittleFS. Doesn't seem like a good plan. I'll close this issue, and if/when ESP32 supports LittleFS, or when ESP8266 ceases support for SPIFFS, we'll deal with it then.

closed 2020-10-12 by ba58smith