SignalK/SensESP #187: Timeout in Data Transmission

kdschmidt1, 2020-10-06:
hello,
trying sensesp with signalk (1.34.0) i observed an interruption of data transmission approx. every 20 seconds (the data in signalk remains for some time (~10sec) constant). i use only the internal uptime as sensor. any idea how to solve?

ba58smith, 2020-10-06:
How solid is your wifi connection from the ESP to the network? I've seen just data lags on my boat from ESP's that are way down in the engine room, which is separated from my wifi router by a steel floor. In my case, I feel like it's a wifi problem because when UPTIME does start to refresh again, it's still accurate. That is, it was 510, then it didn't update for about 10 seconds, and then it was 520. Could you be experiencing the same thing?

kdschmidt1, 2020-10-13:
hello,
my wifi connection is quite solid. all devices inside 5 m^2.
i tested also two different routers, but same result. data logger is showing that all time messages are recieved, but after the timeout in a short period.
Activating DEBUG_ESP_CORE i observed, that during timeout the ack debug message is missing.
do you have any idea how to solve this problem??
following a part of the serial debug messages:

(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"TEST2"},"values":[{"path":"TEST2.uptime","value":77.655}]}]}
␛[0m:wr 98 0
:wrc 98 98 0
:ack 98
(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"TEST2"},"values":[{"path":"TEST2.uptime","value":78.655}]}]}
␛[0m:wr 98 0
:wrc 98 98 0
:ack 98
(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"TEST2"},"values":[{"path":"TEST2.uptime","value":79.655}]}]}
␛[0m:wr 98 0
:wrc 98 98 0
(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"TEST2"},"values":[{"path":"TEST2.uptime","value":80.655}]}]}
␛[0m:wr 98 0
:wrc 98 98 0
(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"TEST2"},"values":[{"path":"TEST2.uptime","value":81.655}]}]}
␛[0m:wr 98 0
:wrc 98 98 0
(D) (get_delta) SKDelta::get_delta: {"updates":[{"source":{"label":"TEST2"},"values":[{"path":"TEST2.uptime","value":82.655}]}]}
␛[0m:wr 98 0
:wrc 98 98 0
:ack 196
:ack 98

ba58smith, 2020-12-17:
No activity for 2 months. Closing.

closed 2020-12-17 by ba58smith