SignalK/SensESP #192: SensESP config page issue

tedenda, 2020-10-07:
A SensESP node with an BME280 sensor attached generates faulty foldout meny on config page.
Humidity value is listed outside the foldout where pressure and temperature is listed:

[image]

The humidity link is not working:

[image]

ba58smith, 2020-10-07:
Please show me the part of your main.cpp where you've created the BME280Value. You may not have the config_path quite right. (Does it start with a "/"?)

tedenda, 2020-10-07:
Good catch!

I will make a PR to fix the example code!

```
  // Do the same for the humidity value.
  auto* bme_humidity =
      new BME280Value(bme280, humidity, read_delay, "Outside/Humidity");

  bme_humidity->connect_to(new SKOutputNumber("environment.outside.humidity"));
```

tedenda, 2020-10-07:
Its now correct:
[image]

I have fixed a PR for this issue:
https://github.com/SignalK/SensESP/pull/193

ba58smith, 2020-10-07:
Nice catch - thanks!

closed 2020-10-07 by ba58smith