SignalK/SensESP #216: Update fx8700_heading_example to write to correct SK Path

tedenda, 2020-10-19:
When I was updating all examples to output valid SK Paths I found that fx8700_heading_example don't format the Roll/Pith/Yaw according to [documentation](https://signalk.org/specification/1.4.0/doc/vesselsBranch.html#vesselsregexpnavigationattitude).

I discussed my finding with @mairas and the conclusion is:
`It should be just one path that outputs a Json object of the given format.`

I don't find any valid SK paths for acceleration and rate so how we should handle those is still an open question.

BjarneBitscrambler, 2020-10-22:
Thanks for bringing this up. I can have a look at this issue in a couple days - I'll wait till PR #166 is closed first.

tedenda, 2020-10-22:
@BjarneBitscrambler I started to code, added the formatting files, but I don't know how to test it nor implement it in NXP FX87. Got some tips from @mairas how to test it via the LambdaTransform for example, but I didn't figure it out. Please continue this work and I will eventually learn a bit.

You find my work at my fork under branch AddAttidudeFormat.

DanielG86, 2020-10-24:
Regarding the SK-paths: I've noticed that it writes to "orientation.XXX", and not to "navigation.XXX". 
Is this the same remark you made, @tedenda ?

tedenda, 2020-10-24:
I'm aware of this but waiting to fix it. When we have fixed that this sensor can output Attitude data correctly we need to update this example anyway. There is other things that also need to be fixed.

I'll fix it very soon.

tedenda, 2020-10-24:
I have now updated the example to match valid SK paths, added attitude that will be added etc.
Work is done in PR #225 But merging that has to wait until this issue is solved and merged.

BjarneBitscrambler, 2020-12-20:
@tedenda and @DanielG86 - I can adjust the paths in the orientation example if you want. I'm editing anyway to incorporate the fusion library v7.  If you want to try out my first pass it's at https://github.com/BjarneBitscrambler/SensESP.git in the OrientationV7 branch. I only focussed on getting it working on the ESP32 - I haven't yet tried it on my ESP8266, nor have I started putting in the various changes that have accumulated in SensESP over the past few months...

tedenda, 2020-12-20:
I have a new ESP32 now so I can try it.

PR #405 (2021-04-07): Remove unnecessary AHRS dependency
Fixes #403.
Fixes #244.
Fixes #216.
commit 86195a5e (2021-04-07): Remove unnecessary AHRS dependency.

Fixes #403.
Fixes #244.
Fixes #216
closed 2021-04-07 by mairas