SignalK/SensESP #242: ADS1x15Voltage transform must be integrated with the sensor itself

mairas, 2020-11-06:
It makes no sense to have a sensor object output raw values which would then unconditionally need to be fed to a transform. The ADS1x15Voltage transform should be integrated to ADS1x15Value.

ba58smith, 2020-11-07:
The reason I did it this way is that one doesn't always use an ADC to determine the voltage of something. You may simply want to know the level of a tank, in which case connecting ADS1x15Value to a Linear transform with the appropriate multiplier and offset will give you what you want.
However, if ADS1x15Value becomes ADS1x15Voltage, you would still need to use a Linear to convert voltage to tank level, so I don't suppose it matters.
I could leave ADS1x15Value as it is, and create a new ADS1x15Voltage. That would (a) not break anyone's existing main.cpp, and (b) give the Class a proper name. ("ADS1x15Value" doesn't sound like a Class that would output a voltage.) Let me know what you'd like, and I'll do it.

mairas, 2020-11-08:
Your proposal sounds great! ADS1x15Value can then be deprecated and removed later.

ba58smith, 2020-11-12:
@mairas, here's another thought. If one assumes that a main reason for using an ADS1x15 is to determine the voltage of something, it would be quite simple to make the new Class `ADS1x15Voltage` output not only the voltage that came to the ADS1x15, but to also be able to output the original voltage that's being measured. Of course, in any case where the voltage being measured is greater than 5.0 V (the highest that the ADS1x15 chips can handle, I think), there's still another Transform that has to be used, the `VoltageMultiplier`, which reverses the effect of a physical voltage divider circuit that's been used to step down the voltage to the ADS1x15's maximum input voltage.

With a couple of parameters (the values of the two resistors used in the voltage divider circuit), `ADS1x15Voltage` could output the originally-measured voltage, without having to use the `VoltageMultiplier` Transform. And if those two parameters are not provided, with defaults being 0 for each, `ADS1x15Voltage` would output whatever voltage came to the physical ADS1x15 chip.

I'll write it either way - let me know your preference.

mairas, 2020-11-13:
I hadn't actually looked at the details that closely. And maybe didn't even think the whole thing out. 🙈 I solved the scaling issue in `AnalogInput` by just adding an `output_scale` parameter that defines the maximum swing of the output. The default is 1024, which is the raw analog input maximum scale in ESP8266 but if you define it to 1.1, you'll get output in volts. Wouldn't a similar approach work perfectly well for ADS1x15 too? The change would be backwards-compatible and more generic than just defining the voltage (which, as you said, might not be what the developer wants in the end).

ba58smith, 2020-11-13:
"The default is 1024, which is the raw analog input maximum scale in ESP8266 but if you define it to 1.1, you'll get output in volts."

You lost me there. If the original input voltage is, say, 2.5V, out of a possible 3.3V. 2.5 / 3.3 = .7576, times 1024 = 776. But .7576 x 1.1 = .83336, which is not at all close to 2.5. But nevermind that. I don't think it's a good way to address my question about using the ADS1x15Voltage Transform to transform the input to the ADS1x15 chip into the ORIGINAL voltage that the user is trying to measure. For example, if you use an ADS1115 to measure the voltage of a 12V battery, you first have to use a physical voltage divider circuit (an R1 and and R2) to step the approx. 12V down to no more than 5V. The way it works now (in https://github.com/SignalK/SensESP/blob/master/examples/ads1x15_volt_meter.cpp) is this:
1. Run 0-12V through a physical voltage divider to get 0-5V.
2. Measure the 0-5V with an ADS1115, which outputs 0-32768.
3. Run that through the ADS1x15Voltage Transform, which outputs 0-5V.
4. Run that through the VoltageMultiplier Transform, which outputs 0-12V.

I will definitely combine steps 2 and 3, to output the voltage that was measured by the ADS1115, in a new Class of Sensor called ADS1x15Voltage. (Leaving the original ADS1x15Value in place.)

I'm saying we could also combine step 4 into the ADS1x15Voltage Transform, as an option, by adding the values of R1 and R2 as optional parameters to the constructor. If they're not provided, you get the voltage as measured by the ADS1115. If they ARE provided, you get the voltage before the voltage divider, thereby eliminating the need using the VoltageMultiplier Transform.

Yes, you could accomplish the same result with a generic `output_scale` parameter, but that assumes that people who are not good at math will have any idea how to calculate that number. My proposal makes it extremely simple for any end user to understand how to use an ADS1x15 chip to get whatever they want out of it:
- ADS1x15Value to get the raw output from the chip.
- ADS1x15Voltage to get the voltage that was read by the chip.
- ADS1x15Voltage with R1 and R2 provided to get the original voltage that's really being measured.

Sorry for the long explanation. I wasn't sure my first explanation accurately conveyed my intent.

ba58smith, 2020-11-29:
Closed with PR #263.

closed 2020-11-29 by ba58smith