SignalK/SensESP #256: TwoWire::begin() has conflicting definitions in ESP32 and ESP8266 versions of Wire library

BjarneBitscrambler, 2020-11-17:
## Problem
The Wire libraries for the ESP32 and ESP8266 differ in an important manner, despite being written by the same author.  The `TwoWire::begin()` method for the ESP32 has prototype:
```
bool TwoWire::begin(int sdaPin, int sclPin, uint32_t frequency);
```
whereas for the ESP8266 it is:
```
void TwoWire::begin(int sda, int scl, uint8_t address);

```
## Impact
The consequence is that when calling TwoWire::begin() for an ESP32 with a reasonable frequency (say 400000 Hz) parameter, it will on the ESP8266 instead be interpreted as starting a connection with address 0x80 (i.e. 0x061a80 is truncated to a uint8_t), at whatever the default frequency which likely will not be what it desired. 

## Fix
One workaround is to split up the `begin()` call into two:  a call to begin() specifying only the SDA and SCL pins, and a second call to `setClock()`.  In this form, both calls have the same meaning in both libraries.

Another workaround (less preferable I'd say) is to have conditional-compilation #ifdef / #endif around two versions of the code.

## History

@ba58smith pointed out (thanks!) a warning seen in when compiling the orientation code for the ESP8266, which I hadn't seen when targetting the ESP32. 
Re: https://github.com/SignalK/SensESP/issues/249#issuecomment-729140276_
```warning: large integer implicitly truncated to unsigned type [-Woverflow]
   Wire.begin(pin_i2c_sda, pin_i2c_scl, 400000 );
```

ba58smith, 2020-11-17:
I like your first solution. Seems cleaner.

BjarneBitscrambler, 2020-11-17:
I had a look at the other sensors using I2C (*SHT31, INA219, BMP280, BME280, ADS1X15*) and none of them appear to try explicitly setting the I2C clock speed using begin(), so I believe they are unaffected by this issue.

Also, just for reference, the versions of the Wire library that are in discussion are: for ESP32 
```name=Wire
version=1.0.1
author=Hristo Gochkov
maintainer=Hristo Gochkov <hristo@espressif.com>
```
and for ESP8266:
```
name=Wire
version=1.0
author=Arduino
maintainer=Ivan Grokhotkov <ivan@esp8266.com>
```
I had earlier said they had the same author (Nicholas Zambetti, 2006), and they have shared some maintainers since then, but somewhere along the way they diverged a bit.  There are a few other differences I can see:
- ESP8266 version has `setClockStretchLimit()` but ESP32 doesn't
- ESP32 version has (and ESP8266 hasn't)  `getClock(), setTimeOut(), getTimeOut(), lastError(), getErrorText(), writeTransmission(), and readTransmission()`

ba58smith, 2020-11-17:
Maybe you can find a library that is the same for both ESP's, and try that?

What happens with your code if you don't explicitly set the clock speed, like with the other I2C sensors?

BjarneBitscrambler, 2020-11-18:
I spent ~20 minutes browsing the libraries (there are a lot!) but didn't find anything that was identical for both ESP32 and ESP8266. So unless someone wants to modify one or both versions, we're stuck with the differences. Both of the ones pulled in by SensESP are the most widely-used and I'd be reluctant to go with a niche library solely to correct a problem that is fairly straightforward to avoid in the higher-level code.

I confirmed that both the ESP8266 and ESP32 (in `core_esp8266_si2c.cpp` and `esp32-hal-i2c.c` respectively) default to 100kHz I2C clock unless one specifies something different.  I prefer using the high speed mode as there is a lot of data coming from the 3 sensors. For example, on the current version 7 fusion code (over on my repository), which has the sensors measuring at 200 Hz and transferring blocks of data at 40 Hz, each I2C block transfer takes about 5 ms. I can't recall what it takes for the version 4 code (currently what SensESP uses), but it would be similar.

PR #258 (2020-11-19): Split frequency from Wire.begin() for ESP8266 compatibility
Issue #256 noted that the `begin()` method of the Wire library differs between the ESP8266 and ESP32 versions. This PR removes the frequency from the begin() call, and instead makes a separate call to `setClock()`  The code is now compatible with both versions of the Wire library and fixes #256.

I've verified that it compiles without warnings for the d1_mini and ESP32-Wrover, and runs on the Wrover. I don't have a board to test the d1_mini executable (but have one ordered).
closed 2020-11-19 by mairas