SignalK/SensESP #262: EspSoftwareSerial compiler failure for ESP8266

ba58smith, 2020-11-21:
I opened the following Issue with the EspSoftwareSerial library:
*****************************
When trying to build a PlatformIO Project that uses a brand new version of EspSoftwareSerial (6.10.0), I get:

"pio/libdeps/d1_mini/EspSoftwareSerial/src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override",

but only for my ESP8266 Project. My ESP32 project builds correctly.

When I revert to version 6.9.0, PlatformIO for some reason uses version 6.8.5, but the Project builds correctly for ESP8266 and ESP32, with no other changes.

Everything was working until version 6.10.0 was released a few days ago.
*******************************
I got this reply to the above Issue: "Don't use the EspSoftwareSerial from the Arduino library repository with the ESP8266, it's already versioned and part of the ESP8266 Arduino core. For ESP32, do use the Arduino library repository, though. Even with the ESP32, be sure to match new EspSoftwareSerial releases to current, often "unreleased", Git master branch commits."

mairas replied: "OK, so it just needs the platform definition in library.json . And if I understood the last sentence, we should also peg the version."

Ianknx, 2020-11-26:
Hello

I am having problems with compile and upload with the following messages:-

oard_build.ldscriptWarning! '-Wl,-T' option for specifying linker scripts is deprecated. Please use 'board_build.ldscript' option in your 'platformio.ini' file.
Retrieving maximum program size .pio\build\d1_mini\firmware.elf
Checking size .pio\build\d1_mini\firmware.elf
*** [.pio\build\d1_mini\firmware.elf] Implicit dependency `C:\Users\Ian\.platformio\platforms\espressif8266\builder\board_build' not found, needed by target `.pio\build\d1_mini\firmware.elf'.

I have followed the install instructions to the letter but am getting nowhere can you please advise.

Regards

Ian

Ianknx, 2020-11-26:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

; NOTE: This is an example platformio.ini file to be used as a template for
; derived projects. Do not use the file within the SensESP directory tree but
; instead create a new project in PlatformIO and copy this file and one of the
; example source files there.

[platformio]
;set default_envs to whichever board(s) you use. Build/Run/etc processes those envs
default_envs = 
   ;esp32dev
   d1_mini
;   esp-wrover-kit

[env]
; Global data for all [env:***]
framework = arduino
lib_ldf_mode = deep
monitor_speed = 115200
lib_deps =
  SignalK/SensESP

[espressif8266_base]
;this section has config items common to all ESP8266 boards
platform = espressif8266
board_build.ldscript = ldscript
build_flags =
   -Wall
   -Wno-reorder

[env:d1_mini]
extends = espressif8266_base
board = d1_mini
board_build.f_cpu = 160000000L
upload_resetmethod = nodemcu
build_flags =
   ${espressif8266_base.build_flags}
   ; Verify that this is the correct pin number for your board!
   -D LED_BUILTIN=2
upload_speed = 460800  
; uncomment and change these if PlatformIO can't auto-detect
; the ports
;upload_port = /dev/tty.SLAB_USBtoUART
;monitor_port = /dev/tty.SLAB_USBtoUART

[espressif32_base]
;this section has config items common to all ESP32 boards
platform = espressif32
build_unflags = -Werror=reorder
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder

[env:esp32dev]
extends = espressif32_base
board = esp32dev
; Verify that this is the correct pin number for your board!
build_flags = 
   -D LED_BUILTIN=2
; uncomment and change these if PlatformIO can't auto-detect
; the ports
;upload_port = /dev/tty.SLAB_USBtoUART
;monitor_port = /dev/tty.SLAB_USBtoUART

[env:esp-wrover-kit]
extends = espressif32_base
board = esp-wrover-kit
upload_speed = 460800

DanielG86, 2020-11-29:
I used the adjusted platformio.ini version of 24 love, but it doen't work for me. The error messages show it fails on missing senses_app.h, and also following included signalK files are missing. 
I looks like the SenseESP libraries are not included in the platformio.ini anymore (lib_deps doesn't include SignalK/SensESP anymore. 
But if I include that, the compiler fails on the EspSoftwareSerial for mu D1-mini.
How could this be resolved?

NB: there are now two platformio.ini files in the repository. One under examples, and one under the main code files.

ba58smith, 2020-11-29:
@DanielG86 - please read the README (and probably the Wiki - there's a reference to it in the README), for instructions on how to get a working `platformio.ini`. (Hint: unless you're developing new code for SensESP, you will use only the one in /examples.)

Ianknx, 2020-11-29:
Daniel

I had a few problems in this area and got things up and running with the
help from great people on Slack.

Hope this helps.

Ian

DanielG86, 2020-11-29:
Thanks for your help. With the following adjustment on the platformio https://github.com/SignalK/SensESP/pull/267#issuecomment-735420399
, I got it to compile again. Thanks @ba58smith !

PR #295 (2020-12-05): Remove EspSoftwareSerial dependency
As discussed in #283, this PR removes the EspSoftwareSerial PR altogether. I refactored the gps_compass example to use SoftwareSerial (which is provided by the Arduino SDK) on ESP8266 and another HardwareSerial UART on ESP32. The resulting #ifdefs are a bit messy but at least the dependency could be removed for good.

Fixes #262.
commit 2c915c8b (2020-12-05): Remove the EspSoftwareSerial dependency

Fixes #262.

@ba58smith, this also supersedes your PR #283.
closed 2020-12-05 by mairas