SignalK/SensESP #274: ENH: Avoid enum namespace collision in sensors.

patg, 2020-11-25:
Using certain sensors together in the same app lead to namespace collisions.
In my case I wanted to use bmp280 and sht31. Both sensors define an enum for their
value initializers and the build fails. Unfortunately the enum is not defined in the class, 
but globally which leads to an error of  redefining 'temperature' for the above example.

In sensors/bmp280.h:
`enum BMP280ValType { temperature, pressure };`

A fix would be to move the enum into the class definition:
```
class BMP280Value : public NumericSensor {
 public:
// Pass one of these in the constructor to BMP280value() to tell which type of`
  // value you want to output
  enum ValType { temperature, pressure };
 BMP280Value(BMP280* bmp280, ValType val_type, uint read_delay = 500,
              String config_path = "");
.....

private:
   ValType val_type;

```

the usage will slightly change, i.e the initialization has to be changed from

`auto* temp =new BMP280Value(bmp280, temperature, read_delay, "/Outside/Temperature");`
to
`auto* temp =new BMP280Value(bmp280, BMP280Value::temperature, read_delay, "/Outside/Temperature");`

mairas, 2020-11-25:
Right, makes sense. The enumerator names are a bit broad for global identifiers, anyway. Your solution works, or alternatively they could be wrapped into an `enum class` which is pretty much just another way of doing the same thing. I learned some new things from these links:

https://stackoverflow.com/questions/4269365/do-enum-values-behave-like-global-variables
https://en.cppreference.com/w/cpp/language/enum

Do you know how to do a PR? If so, we could get the fix in quickly; otherwise it might have to wait for a couple of days.

ba58smith, 2020-11-25:
That was my mistake, and I have time to do the PR now, so I will.

ba58smith, 2020-11-26:
Closed with PR #275 .

closed 2020-11-26 by ba58smith