SignalK/SensESP #301: LedController should be renamed

mairas, 2020-12-07:
LedController is the controller class for the system status Led (ordinarily the onboard LED on GPIO 2). The class name is too specific and should be renamed to something like StatusLedController.

joelkoz, 2020-12-07:
I took a peek at these two classes.  My suggestions:

1. Rename `VisualOutputController` to `SystemStatusConsumer`
IMO, this class is really like an interface definition, and its primary purpose is to consume input regarding the network status. There might be a number of other purposes for doing this - not just creating a visual indicator. Given this new name, we may want to move it out of the `controller` directory and into the `system` directory

2. Rename `LedController` to `SystemStatusLed` as it represents a single led that outputs the system status.

3. We might benefit from breaking the implementation of this new`SystemStatusLed` into a base `SystemStatusController` that holds the implementation of the `set_input()` methods and defines the  `set_wifi_*()` and `set_ws_*()` methods as abstract (or better yet, give them default empty implementations).  Again, responding to the system status for things other than a visual indicator might be beneficial.  That interim class would be much more in line with the definition of a "controller" IMO, and deserves to be placed in the controller directory.

4. Refactor `SystemStatusController` to be a `ValueProducer<SystemStatusValue>`
To really leverage this work, the other possibility would be to make `SystemStatusController` also be a `ValueProducer` of events, similar to the way `ClickTypes` was implemented.  By doing that we'd have network events consumed by a SystemStatusController (a specialized version of `SystemStatusConsumer`), translated into a system event enumeration value then output, and then `SystemStatusLed` could be implemented as a simple `ValueConsumer<SystemStatusValue>` of those enumerations.

Finally if the `SystemStatusController` were a singleton (explicitly designed or implied by being a globally available member of `sensesp_app`), then its output could be connected to more than one entity that might be interested in these status values (like the SKDelta that needed to reset its meta flag when the system re-connected to the server), or any user code that needs to react to the server being present (or not).

mairas, 2020-12-08:
Good proposals. I'll try using them as a guideline and see what happens.

PR #310 (2020-12-11): Refactor SystemStatusController into ValueProducer
`SystemStatusController` has been so far a `ValueConsumer` for `WSConnectionState` and `WifiState` and the use model was to inherit it to define actual behavior. This PR refactors `SystemStatusController` into a ValueProducer, permitting users to simply consume the `SystemStatus` updates instead.

The change has also been implemented in SensESPStatusDisplay.

Fixes #301.
closed 2020-12-11 by mairas