SignalK/SensESP #314: Token renewal issue after SensESP is deleted from SK server

KEGustafsson, 2020-12-11:
SensESP cannot renew access token after it is removed from SK server.

`(D) (test_token)(C1) End of payload output
(D) (test_token)(C1) Attempting to connect to Signal K Websocket...
(I) (connect)(C1) WSClient websocket connect attempt (state=0)
(D) (connect)(C1) Initiating websocket connection with server...
(D) (connect)(C1) Websocket is connecting to Signal K server on address 192.168.0.2:3000
(I) (connect)(C1) WSClient websocket connect attempt (state=0)
(D) (connect)(C1) Initiating websocket connection with server...
(D) (connect)(C1) Websocket is connecting to Signal K server on address 192.168.0.2:3000
(I) (connect)(C1) WSClient websocket connect attempt (state=0)
(D) (connect)(C1) Initiating websocket connection with server...
(D) (connect)(C1) Websocket is connecting to Signal K server on address 192.168.0.2:3000
(I) (connect)(C1) WSClient websocket connect attempt (state=0)`

Reflashing ESP cannot cure the situation, but chip need to be erased before reflashing.

mairas, 2020-12-11:
@joelkoz could this be a corner case left out from PR #309?

joelkoz, 2020-12-11:
Indeed it could.  One question: Is there a reason that the schema returned from `get_config_schema()` of `ws_client.cpp` has the token marked as `readOnly: true`?  If that were allows to be re-written, the quick solution would be to use the config gui to simply blank out that field.

joelkoz, 2020-12-11:
Actually, upon review, we have this code that looks at the return of the token validation:

ws_client.cpp line 351:
```
    } else if (httpCode == 401) {
      this->client_id = "";
      this->send_access_request(server_address, server_port);
    } else {
```

I'll run a quick test and see what the return value is.  If the server rejects with a 401 (invalid access), it should clear the token.  If it returns something like 403 (forbidden), we are missing that.

joelkoz, 2020-12-11:
I just tested this.  When I first deleted the token, I got the repeating "WebSocket attempt" over and over as reported.  However, when I reset the device, I am now getting the correct "PENDING". Authorizing the token makes it work.

So - the solution is to reset the device after deleting the token.  At least, this works for the node server version of SignalK server.

mairas, 2020-12-11:
That doesn't sound great. The devices might be in hard-to-reach places or have some annoying configuration to be done. Also, this same issue would likely happen if a time-limited token would expire. Shouldn't we just delete the token for any 401 or 403 response?

KEGustafsson, 2020-12-11:
It seems that token validation returns 200.
It should be 403 according to [After Access Approval](https://signalk.org/specification/1.5.0/doc/access_requests.html#after-access-approval)

joelkoz, 2020-12-11:
@KEGustafsson - do you have "Allow read only access" turned OFF on your Signal K server?  You need to do that.  SensESP apps are almost always a "write" device in some form or another (i.e. it sends data TO the server, vs. just reading it and reacting).  The token validation code assumes if security is active, read only access is forbidden.

joelkoz, 2020-12-11:
Also for clarification for anyone reading this, the real solution to this problem is "if you manually delete a token from the SignalK server, you need to reboot the device."  That is, all you need to do is use any web browser to connect to the SensESP device (as if you were going to configure it), and select the "Restart device" option.  That will cause SensESP to re-test the token for validity (vs. going into a holding pattern thinking the server has gone down).

joelkoz, 2020-12-11:
Also note that after extensive research on this, an unfortunate "feature" of the web sockets library we are using is that it does not pass along the http status codes to users of the library when it makes its callbacks.  The "unauthorized" response, and the "connection refused because the server is down" response looks identical to our code.  There is no way to get this http status because it is cleared before the callback is made. Couple this with the fact that the SignalK server does not return a Json error when not authorized and relies only on the Http return code when rejecting connections makes it impossible for us to detect this situation automatically without forcing the token to be tested on each re-connect.  There are some reasons we don't want to do that. We'll have to discuss if those reasons are good enough to take the drastic steps that would be required to fix this corner case.

KEGustafsson, 2020-12-12:
About turning off "Allow read only access". Can this lead to issues with other application, which are using ws without token?
Could there be a counter for initiating websocket? If too many tries, then reboot. This would require turning off read only access or can it be skipped if forced reboot for too many unsuccessful tries?

joelkoz, 2020-12-12:
After some internal discussion with the dev team, we came to this simple but effective compromise fix: #321

KEGustafsson, 2020-12-13:
Tested your compromise and so far it is working OK.
Anyway, `Readonly` caused other issues, so therefore I raised an issue in server side, [Status of token #1216](https://github.com/SignalK/signalk-server/issues/1216)

joelkoz, 2020-12-15:
Fixed by #321

closed 2020-12-15 by joelkoz