SignalK/SensESP #315: ENH: Overload I2C constructors to pass alternate wiring configurations to Adafruit libraries

cptnslick, 2020-12-11:
Sorry for the long post. I'm new to this. Not sure how to Pull...
Summary:
I needed to get I2C working on non-standard pins. Platform - ESP32 Pico. I called Wire.begin(SCL, SDA, FRQ) in Main.cpp and modified the constructor in the BMP280 sensor library to pass it on the Adafruit library. It would be great to overload the constructors in the rest of the I2C compatible libraries to offer this ability. 

Look for "Wire" and "theWire" add-ins

----Main.cpp----
//define SCL & SDA pins, Frq = 100000
 Wire.begin(26, 32, 100000);

// Create a BMP280, which represents the physical sensor.
// 0x77 is the default address. Some chips use 0x76, which is shown here.
auto *bmp280 = new BMP280(0x76, &Wire);

----BMP280.h----
class BMP280 : public Sensor {
 public:
  BMP280(uint8_t addr = 0x77, TwoWire *theWire = &Wire, String config_path = "");
  Adafruit_BMP280* adafruit_bmp280;

 private:
  uint8_t addr;
};

----BMP280.cpp----
BMP280::BMP280(uint8_t addr, TwoWire *theWire, String config_path)
    : Sensor(config_path), addr{addr} {
  load_configuration();
  adafruit_bmp280 = new Adafruit_BMP280(theWire);
  if (!adafruit_bmp280->begin(addr)) {
    debugE("Could not find a valid BMP280 sensor: check address and wiring");
  }
}

ba58smith, 2020-12-11:
I will have a look at this as soon as I can. Shouldn’t be too tough.

ba58smith, 2020-12-12:
See PR #319

cptnslick, 2020-12-12:
Thank you! This is an awesome project. Keep up the great work.

ba58smith, 2020-12-13:
@cptnslick, notice that I've changed the implementation. I ended up making no changes at all to the example, so to see how to use the new version of the constructor, look at the comments in the .h file.

ba58smith, 2020-12-14:
@cptnslick , if you can test this PR, I would appreciate it. I don't have a BMP280. In your project's `platformio.ini`, where it now says `lib_deps = SignalK/SensESP` (or some other reference to SensESP), change it to `lib_deps = https://github.com/ba58smith/SensESP/tree/BMP280Update`. That will pull the code from the BMP280Update branch on my fork. Then, exit PlatformIO, delete your `/PlatformIO/Projects/YourProject/.pio/` directory, and go back into PlatformIO. Wait until the "Intellisense Index Rebuild" message in the blue status bar along the bottom finally stop, then build the project, upload it to your ESP, and test it.

cptnslick, 2020-12-14:
Good Morning,
  I followed your steps. Looks like a few libraries are missing. That is
what I'm seeing.

[image: image.png]

ba58smith, 2020-12-14:
@cptnslick - I don't think that was the image you wanted to uploade. (BTW, you can copy - paste text from your screen to Slack, by putting three "backwards apostrophes" - the character on the same key as the tilde ( ~ ) on many keyboards - before and after the text. That sets it off here in Slack.)

ba58smith, 2020-12-17:
@pctnslick, this has been merged to SensESP/master, and will be part of a soon-to-be-released Version 1.0. When it is released, change your Project's `platformio.ini` so its `lib_deps = ` points to `SignalK/SensESP`. Or, you can always point it to `https://github.com/SignalK/SensESP`, which will get you the latest merged code, which may or may not be in the official "Release" version.

cptnslick, 2020-12-17:
Thank you! I tested it out this evening. Works great.

ba58smith, 2020-12-17:
Closed with PR #319.

closed 2020-12-17 by ba58smith