SignalK/SensESP #334: ENH: Modularize SensESP protocols and communications layer

joelkoz, 2020-12-18:
This is a place to make notes on the steps we would need to take to modularize SensESP to allow SignalK to be but one of many protocols SensESP speaks, and for Websockets to be but one of many communication layers.  Initial steps in this refactor:

1. `WSClient` and `Networking` class needs to inherit from `Enable`
2. Explicit construction and enabling of those classes needs to be removed from `SensESPApp` and moved into `SensESPAppBuilder`
3. `WSClient` needs to be refactored to move the websocket specific code into a base class and separate out the `SignalK over websocket` code into another class. Ideally, these would both be `SymmetricTransform<String>` to decouple SignalK processing from networking layer.
4. The delta sending/receiving coordination between`SKDelta`, `SKOutput`, `SKPutRequest`, `SKPutRequestListener` and `SKListener` needs to be refactored to use whatever new communication layer mechanism arrived on in 3

mairas, 2020-12-18:
`SKEmitter` would need to be generalized to apply to any type of output SensESP might produce, e.g. NMEA 2000 PGNs or NMEA 0183 sentences.

Likewise for SKOutput. They're at the moment inherited but maybe the emitters and transport outputs should be separate beasts in the output chain? We might have SK deltas over websockets, plain TCP pipe, or RS-485. Or NMEA-0183 sentences over RS-422, or UDP broadcasts.

WSConnectionState could be generalized to something like AppConnectionState that would then apply to websockets, direct TCP streams, or (with a bit of shoehorning) to UDP broadcasts.

Likewise, WifiConnectionState would be NetworkConnectionState and would be applied to wired ethernet, CAN bus, NMEA 0183 (RS-422) and RS-485 as well.

kupo2kupo, 2021-03-25:
Would this be a required change to make RS485 communication with sensors work? I am now using RS485 submersible pressure sensors as tank sensors, very accurate and robust. Currently I use an RS485-USB interface to read the data into a RPi, and forward the data into Influx. But a dedicated solution based on SensESP would make more sense.
Or is there already another way to make RS485 work with SensESP?

mairas, 2021-03-25:
I'd say it's not required. It'd make things cleaner, for sure, but definitely not required. You can write producers and consumers that implement your RS-485 communication. They could be either proper RS-485 classses, or you could just instantiate ObservableValues and update them on the producer side and have LambdaConsumers on the consumer side.

I don't know anything about Arduino framework libraries for RS-485, so I'll use NMEA 2000 as an example. Have a look at my SH-ESP32 test jig software at https://github.com/hatlabs/SH-ESP32-test-jig. The controller uses SensESP producers and consumers (but does not actually instantiate SensESPApp, so no Signal K or WiFi is used) while the device under test software just uses the ReactESP events for parsing input and writing data on the bus.

You can ask me for more guidance on the SK Slack if you wish (or email me at mairas@iki.fi).

kupo2kupo, 2021-03-30:
Thanks for the response. There is an Arduino Modbus library (https://www.arduino.cc/en/ArduinoModbus/ArduinoModbus) which I will try. But I will first try to get my multi-sensor Modbus installation running on the PI before trying to port to SensESP.

ruedigerO, 2023-01-04:
Is there any plan / work in progress to implement this?

For me it would be a great benefit for a project I like to do.

To save energy I would like to build a panel with a touch display, relay board and some sensors (BME, tank, etc.) programming based on SensESP. The panel should work stand alone (no WiFi, no Raspi) to save power but if the Raspi with Signal-K comes up the panel should also deliver the sensor data to SK.

From my point of view the best way for this is to send SK via serial.