SignalK/SensESP #355: Can not open SenseESP.gzz

norbert-walter, 2021-01-06:
I can not open the SenseESP.gzz with Fritzing V0.9.2 under Linux. Is it possible to make the schematic as PDF file? It is helpfully for viewing.

ba58smith, 2021-01-06:
I had no trouble opening it. I did get a notice that one component couldn't be found, but then it opened.

This isn't really a part of SensESP - it's a PCB that one user created to allow someone to use a Wemos D1 Mini with several different types of sensors connected. If you want one, the information to order one is there in the same folder in the repo as the .fzz file. If you just want to see what it looks like, there's a photo of it also in the same folder.

norbert-walter, 2021-01-07:
Thanks for the answare. I need a schematic with wires for all components. The picture in the same folder are not helpfully.

ba58smith, 2021-01-07:
[image]

ba58smith, 2021-01-07:
[image]

ba58smith, 2021-01-07:
[image]

closed 2021-01-07 by ba58smith

norbert-walter, 2021-01-07:
Thanks for all