SignalK/SensESP #369: Increase max rate of SK updates

BjarneBitscrambler, 2021-02-05:
Currently, the max rate at which SK updates are sent to the server is 10 Hz.  This value is set in `net/ws_client.cpp`:
```
void WSClient::enable() {
  app.onDelay(0, [this]() { this->connect(); });
  app.onRepeat(20, [this]() { this->loop(); });
  app.onRepeat(100, [this]() { this->send_delta(); });          <<< throttled to max 10 Hz
  app.onRepeat(10000, [this]() { this->connect_loop(); });
}
```
If a fast sensor (e.g. GPS or compass) generates updates at a higher rate, they are stored and sent out in batches at 100 ms intervals. Here, for example, is a typical delta sent out at 100 ms intervals when the attitude and heading reports intervals are set to 50 ms:
```
(D) (get_delta) delta: {"updates":[{"source":{"label":"SensESP_D1"},"values":[
     {"path":"navigation.attitude","value":{"yaw":2.853591,"pitch":3.072033,"roll":1.368445}},
     {"path":"navigation.headingCompass","value":2.853591},
     {"path":"navigation.headingMagnetic","value":2.940857},
     {"path":"navigation.attitude","value":{"yaw":2.853448,"pitch":3.071935,"roll":1.368391}},
     {"path":"navigation.headingCompass","value":2.853448},
     {"path":"navigation.headingMagnetic","value":2.940714}
]}]}
```
Note the duplicate keys in the delta, due to queued updates.

While 10 Hz is a reasonable update rate for most parameters, responsiveness by controllers such as autopilots and radar plotters can benefit by faster rates. Bear in mind that there are multiple delay points in the signal chain - the cumulative latency between the instant that a parameter is generated and when it arrives at its destination could approach 150 ms or more.

The ESP8266 and ESP32 hardware supports faster update rates. I've been testing both platforms with the change to `app.onRepeat(1, [this]() { this->send_delta(); });`  (i.e. SensESP checks every 1 ms whether a delta is available, and sends it if so) and it has been functioning well. There is even a slight benefit to memory usage (related issue #239) in that deltas that are sent out when available rather than queuing them up will take less RAM.

I propose **changing the interval between `send_delta()` calls to, say, 5 ms.**  This would reduce the latency contribution due to SensESP into the same order of magnitude as that due to other delays, like WiFi router routing and the Signal K server running on a raspberry Pi.

Any reasons not to do this?  I can raise a PR if it sounds reasonable.

mairas, 2021-02-05:
I think this would be a good change. 10 Hz is obviously too slow. The minor downside would be that pretty much every delta would be sent on its own while slower cadences would group them together. I'm not sure this is really an issue, though. The bandwidth increase is minimal and I don't really see any other immediate impacts.

This could also be a configurable value (e.g. via SensESPAppBuilder), but I'm not sure it's even worth that effort; just changing the default should be fine. Go for it!

BjarneBitscrambler, 2021-02-05:
Good point about the deltas being sent individually and the extra overhead, as opposed to when they are grouped.  I think 5 ms is probably a good collection interval:  when I was testing at 1, 2, and 5 ms intervals I noticed that the standard sensors (uptime, freemem, etc) were often split into two updates when using the 1ms setting, and sometimes split when using 2 ms.  At 5 ms, the standard sensors were grouped together in the same delta the majority of the time.

OK, I'll submit a PR later today.

mairas, 2021-02-05:
Thanks!

It's interesting that the standard sensor value output would take a significant portion of a millisecond. I guess the generation of those Json documents is quite expensive...

PR #370 (2021-02-06): Change min interval for sending deltas to 5 ms from 100 ms
Current minimum interval between SK delta messages is 100 ms. This change improves the maximum possible rate of deltas by decreasing the interval between calls to `send_delata()` to 5ms.

This  resolves #369
closed 2021-02-06 by mairas