SignalK/SensESP #377: Current 1-wire implementation doesn't work well on ESP32

mairas, 2021-02-18:
The current 1-wire temperature sensors utilizing the OneWire and DallasTemperature libraries doesn't work reliably on ESP32. The libraries use bit-banging to implement the protocol, but the implementation is such that the RTOS context switching (switching between different tasks - think of mini-processes or programs on bigger OSs) messes up the timings. It works fine as long as WiFi is not enabled, but once there's a lot of background activity, temperature reading fails more often than not.

One possible solution would be adapting the [esp32-owb](https://github.com/DavidAntliff/esp32-owb) library and the related DS18B20 library for SensESP use.

JohnySeven, 2021-02-19:
@mairas have one OneWire device running with SensESP on ESP32 and it works fine. Could you please elaborate what's wrong with Onewire? Thanks

mairas, 2021-02-20:
Plenty of timing errors, resulting in failed readings (also, our current implementation doesn't detect the error codes and passes them forward as garbage values, but that's another issue). It's a known problem. While googling for an example to add, I found this long thread with some good analysis: https://github.com/espressif/arduino-esp32/issues/1335 I had tried adding mutexes around the OneWire calls but didn't get much improvement, but following that thread's ideas might provide some improvement.

pstolarz, 2021-09-06:
@mairas Do you still experience the described timing issues for ESP32 with OneWireNg? Starting from v0.8.1 I added [the code](https://github.com/pstolarz/OneWireNg/blob/501de4bd374748643e58019d24715bd6b722a596/src/platform/Platform_TimeCritical.h#L35) which disables local interrupts for an ESP32 core handling the time critical sections of 1-wire activities. It should fix the issue.

mairas, 2021-09-06:
This issue was created before switching to OneWireNg, I think. I don't think I've ever had those issues with OneWireNg, even before 0.8.1.

I'll close this issue.

mairas, 2021-09-06:
Fixed in #399.

closed 2021-09-06 by mairas