SignalK/SensESP #382: Can't set curve config

ruedigerO, 2021-03-02:
I used the example temperature_sender.cpp and try to configure CurveInterpolator via Web UI but always get an error 
"Error retrieving configuration /12V_alternator/temp/curve: Cannot convert undefined or null to object"

ruedigerO, 2021-03-07:
Short update:
I tested it with a cloned version of the repository and this works!
Did you change anything regarding this functionality?

ba58smith, 2021-03-07:
No, the CurveInterpolator code hasn't changed in a very long time.

ba58smith, 2021-03-07:
How did you test it the first time, if not with a cloned version of the repo?

Regardless of your answer, should we close this Issue?

ruedigerO, 2021-03-07:
As I am not an experienced C++ programmer, I used the steps described in "Getting started" (install VC, create a  platformio.ini, create a main.cpp and start the build). Everything was running fine with exception of the possibility to change settings of the CurveInterpolator via UI. 
In my second test I cloned it via "git clone" and I was able to compile and then change settings. 
Maybe there are changes at the UI/editor which are not committed?

You may close this issue now, but I think for people trying it in the future it would be helpful if it compiles from scratch.

ba58smith, 2021-03-08:
I should have asked this when you first posted the issue: where did you see the error? On the browser screen, or in the Visual Studio serial monitor? That will help me figure out where the error is coming from. And if you have a screen shot of it, with anything else before it, that would be wonderful. Thanks.

ruedigerO, 2021-03-08:
The issue is on the browser if you try to configure the curve.
[image]
[image]

ruedigerO, 2021-04-14:
I think you may close this issue.
Maybe I used an old version of SensESP because now (with very. 1.0.5) it works fine.

Thanks & regards

closed 2021-04-15 by ba58smith