SignalK/SensESP #410: Device IP address 255.255.255.255

Damiasroca, 2021-04-11:
On a NodeMCUv3 with ESP8266 12F 
The device get that IP everytime I try.

Checked on 2 different networks and with 4 different NodeMCU's

```
(I) (wifi_station_connected) Connected to wifi, SSID: Apartament Fora Port (signal: -68)
␛[0m(I) (wifi_station_connected) IP address of Device: 255.255.255.255
␛[0m(I) (connect) WSClient websocket connect attempt (state=0)  
␛[0m(D) (connect) Initiating websocket connection with server...
␛[0m(D) (connect) Websocket is connecting to Signal K server on address 192.168.1.81:3000
␛[0m(D) (connect) No prior authorization token present.
␛[0m(D) (send_access_request) Preparing a new access request
␛[0m(W) (send_access_request) Can't handle response -1 to access request.
```

Any ideas? #

Damiasroca, 2021-04-11:
Digging a little bit more on that issue, I found that in /esp8266/IPAddress.cpp there is that line:
`const IPAddress INADDR_NONE(255,255,255,255);`
So I guess that NodeMCU connects to wifi but it doesn't stay connected?

mairas, 2021-04-11:
To me the result would, together with your INADDR_NONE finding, indicate that the device is able to connect to the wifi but doesn't get an IP address from the DHCP server.

The ESP8266WiFi library does support setting a static IP address using the `WiFi.config()` method but that is not currently exposed to the SensESP API.

Damiasroca, 2021-04-11:
I was testing it with the BME280 example, is the only sensor I have right now at home, adding a  1 second delay right after creating the **sensesp_app** object did the trick. Feels like the NodeMCU need some time for getting the IP from the DHCP server. 
`
...
ReactESP app([]() {
#ifndef SERIAL_DEBUG_DISABLED
  SetupSerialDebug(115200);
#endif

  sensesp_app = new SensESPApp();

  delay(1000);
  // Create a BME280, which represents the physical sensor.
  // 0x77 is the default address. Some chips use 0x76, which is shown here.
  auto* bme280 = new BME280(0x76);
...`

Solved and closing

closed 2021-04-11 by Damiasroca

tedenda, 2021-04-11:
Is it really solved, other users will eventually experience the same thing. My guess is that something needs to be tweaked under the hood so it behave consistent on different hardware.

mairas, 2021-04-12:
It is not a SensESP problem to begin with.