SignalK/SensESP #415: CurveInterpolator Config UI only works up to 10 samples

jcaprano, 2021-04-20:
Hi all,

I have found an issue where the Config UI of CurveInterpolators only works up to 10 samples. Whenever I add more than 10 samples and try to access config of the transform the following error appears:

[image]
[image]

I've tried to do some analysis of the error and found the following:
- It does not matter if the samples are populated 1) by passing a set of samples to the CurveInterpolator constructor 2) Adding samples after constructing the CurveInterpolator through add_samples 3) adding them through the Config UI. Find the part of my main.cpp I used for testing this below:

`std::set<CurveInterpolator::Sample>* samples = new std::set<CurveInterpolator::Sample>;
                samples->insert(CurveInterpolator::Sample(0., (0.0/29)));
                samples->insert(CurveInterpolator::Sample(18., (1.4/29)));
                samples->insert(CurveInterpolator::Sample(34., (2.5/29)));
                samples->insert(CurveInterpolator::Sample(50., (4.3/29)));
                samples->insert(CurveInterpolator::Sample(63., (5.9/29)));
                samples->insert(CurveInterpolator::Sample(73., (8.0/29)));
                samples->insert(CurveInterpolator::Sample(83., (10.2/29)));
                samples->insert(CurveInterpolator::Sample(93., (12.2/29)));
                samples->insert(CurveInterpolator::Sample(103., (14.0/29)));
                samples->insert(CurveInterpolator::Sample(113., (16.5/29)));
                samples->insert(CurveInterpolator::Sample(123., (18.5/29)));
                /*samples->insert(CurveInterpolator::Sample(133., (20.6/29)));
                samples->insert(CurveInterpolator::Sample(146., (22.8/29)));
                samples->insert(CurveInterpolator::Sample(162., (24.8/29)));
                samples->insert(CurveInterpolator::Sample(178., (27.0/29)));
                samples->insert(CurveInterpolator::Sample(194., (29.0/29)));
                samples->insert(CurveInterpolator::Sample(210., -1.0));*/

  auto* freshWater = new CurveInterpolator(samples,"/tankSensor/CurveInterpolator");`
  
- Using the original CurveInterpolator or a modified (self-written) Subclass does not change this behavior

I'd love to help in fixing this error if anyone can point me in the right direction. I find my way around the C++ parts off SensESP but have no idea of anything related to the Config UI..

Cheers,
Johannes

ba58smith, 2021-04-20:
@jcaprano , thanks for spending the time to detail this issue. I will see if I can get @joelkoz (who wrote CurveInterpolator) to have a look. (I looked, but I don't see any obvious limitation of 10 samples.)

ruedigerO, 2021-05-05:
@ba58smith: the same behavior as issue #382
I tried with more than 10 samples and get the same result in the GUI

Regards,
Rüdiger

ba58smith, 2021-05-05:
@jcaprano and @ruedigerO - if you need more than 10 samples, just add them to your `main.cpp`. You don't NEED to add them, or modify them, through the Config UI. That should be an acceptable work-around.

mairas, 2021-05-07:
@jcaprano if you want to debug the issue, have a look at the ArduinoJson buffer sizes. SensESP creates a Json object that defines the config UI. That Json object might take too much space to fit in the buffer that's reserved for it. Or alternatively, if this is on ESP8266, you might be running out of memory.

BjarneBitscrambler, 2021-08-15:
@jcaprano:  @mairas is correct - the JSON objects used in the HTTP PUT and GET methods aren't large enough for more samples. I ran across the same problem yesterday and adjusted the limits to allow greater than 37 interpolation samples. If you want to make the same changes, look in the files:
*  ```http.cpp```  
 #define DYNAMIC_JSON_DOCUMENT_SIZE 2048  //was 1024
 DynamicJsonDocument jsonDoc(3072);  //was 1024
*  ```configurable.cpp``` 
 DynamicJsonDocument jsonDoc(2048);  //was 1024, in two places

The jsonDoc() argument is the max size the JSON object can grow to. Calls to the JSON library's createNestedObject() method fail when adding a new value would result in a size beyond the one specified.

My changes are in https://github.com/BjarneBitscrambler/SensESP/commit/59348a65a8ee113cd189de25ba2160a963bc7cb7 
@mairas - do you want me to open a PR to incorporate this change?

mairas, 2022-02-16:
I had missed Bjarne's comment back in August. Sorry about that!

It looks like a simple fix. Given that the surrounding code has changed quite a bit, I can take the respective changes and make a PR for them. (Unless someone beats me to it - wouldn't mind the least!)