SignalK/SensESP #423: Onewire Temp stop working - "Sensors already configured" please help...

crustulumtheoneandonly, 2021-05-12:
_First of all, I am a beginner, sorry for dump mistakes or annoying questions._

Short Question:

**How to delete the Sensor-Config?**   (uploading the code again wont do)

_Backstory:_

A few days ago is set up a Wemos D1 Mini with 4 Dallas Onewire Temperature Sensors (just like the example project)
Software worked fine so far, I was able to configer der MCU via Wifi AP to connect to my Network and it is sending SystemInfo Data to my Signalk Server. (IP-Adress, SystemHz, Uptime, WifiSignal)

When I connected 4 Sensors (the same as in the example project) they kind of worked for a short period of time.

The I wanted to customize a sigkpath to math the nmea standards. This worked more or less to, but only after I switched the sensor data cables to another pin.

So now, Im not getting any temp sensor data anymore.

When I watch the Wemos D1 boot via serial monitor there is following FATAL error:

FATAL: Unable to allocate a OneWire sensor for /coolantTemperature/oneWire. All sensors have already been configured. Check th␛[0me physical wiring of your sensors.

 
I want to know, how to delete the configuration of the Sensors, because when I open the IP-Adress of the Wemos, I can restart, reset and look at the Informations, but the Configuration Link is broken.

I googled alot about erase flash and also uploaded different adruino and platformio sketches on the MCU, but I cant find any way to reset the Wemos D1 so it will detect the onewire sensors again??

Its been 4 days of trying and I really dont know what to do anymore.

Anyone out ther who can point me in the right direction?

sincerely
a sailor

ba58smith, 2021-05-13:
This is not a good place for this kind of troubleshooting. Please join the free signalk-dev.slack.com, and post your issue in the #sensors channel. You will get good help there. (But the FATAL error you're getting indicates to me that you have a bad physical connection, or you're not connected to the pin that you have configured in main.cpp. Don't reply to that here, though - I won't reply here, but I will on Slack. Thank you.)

crustulumtheoneandonly, 2021-05-14:
Thanks a lot! I will move my question there!

closed 2021-05-14 by crustulumtheoneandonly

crustulumtheoneandonly, 2021-05-15:
@ba58smith I am having trouble joining your slack channel. I guess I dont have the rights to join the workspace?
I never used slack before, so sorry for this questions, I hope Im not annoying. (Cant find any way to direct message you and dont want to post my email here..) Any Idea? (already googled but cant find a way to join your slack without permissions)

tedenda, 2021-05-15:
Hello!

Go to this address and fill in your email and you are up and running.

http://slack-invite.signalk.org

crustulumtheoneandonly, 2021-05-15:
> Hello!
> 
> Go to this address and fill in your email and you are up and running.
> 
> http://slack-invite.signalk.org

Thanks a lot! Such a beginner friendly community!! :)