SignalK/SensESP #424: pwmoutput error

patrol420, 2021-06-20:
im trying for the life of me to get this uplaoded to a wemos d1 mini but i keep getting a 
.pio\libdeps\d1_mini\SensESP\src\system\pwm_output.cpp:61:30: error: 'PWMRANGE' was not declared in this scope; did you mean 'ERANGE'?
   61 |     int output_val = value * PWMRANGE;
      |                              ^~~~~~~~
      |                              ERANGE
error. anyone know what that is

ba58smith, 2021-06-20:
I see that PWMRANGE is defined only for ESP32 (on line 12 of that file), but the line that's throwing the error isn't qualified (with an #ifdef ESP32) to be only for ESP32, so for a D1 Mini (an ESP8266), it's undefined. I think the PWM range for an ESP8266 is 1024, so try this: add these three lines at line 14 of that file:
`#ifdef ESP8266
#define PWMRANGE 1024
#endif`

mairas, 2021-06-30:
The PWM generator of Espressif 8266 Arduino Core has been reimplemented in release 3.0.0: https://github.com/esp8266/Arduino/releases/tag/3.0.1

Apparently, that change removed the definition of `PWMRANGE`. To fix this, `PWMRANGE` should be defined to 1023 which was the previous default value.

patrol420, 2021-06-30:
> I see that PWMRANGE is defined only for ESP32 (on line 12 of that file), but the line that's throwing the error isn't qualified (with an #ifdef ESP32) to be only for ESP32, so for a D1 Mini (an ESP8266), it's undefined. I think the PWM range for an ESP8266 is 1024, so try this: add these three lines at line 14 of that file:
> 
> ```
> #ifdef ESP8266
> #define PWMRANGE 1024
> #endif
> ```
> 
> Let me know if that works, and if it does, I'll make a PR to change it.

thanks mate that worked perfect

mairas, 2021-06-30:
I'm reopening because the issue isn't fixed in the code. And note that the correct value is 1023.

ba58smith, 2021-06-30:
Thanks, @mairas - I'll make the PR.

ba58smith, 2021-06-30:
See PR #426.

closed 2021-07-04 by ba58smith