SignalK/SensESP #425: wont connect to wifi

patrol420, 2021-06-30:
a new issue ive got it flashed ive tryied a NodeMCU and WEMOS D1 MINI, i can get onto the wifi config page but it wont connect to wifi at all, i have 2 access points to choose from and neither board will connect to either. seems to try to connect then just keeps failing

*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
(I) (wifi_station_disconnected) Disconnected from wifi.
␛[0m(I) (wifi_station_disconnected) Disconnected from wifi.
␛[0m

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000c4e20
~ld

patrol420, 2021-06-30:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/sk
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/wifi_reboot_watchdog
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/analog_in
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/linear
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:

ba58smith, 2021-06-30:
@patrol420 , can we take this to the #sensors channel on Slack? Since so many other users ARE able to connect to wifi, I don't know that it's a bug in SensESP. More likely it's something with your configuration, which we can explore better on Slack.

ba58smith, 2021-07-04:
No more info from @patrol420 - closing this issue, as it doesn't seem like a bug, but rather, some kind of configuration issue.

closed 2021-07-04 by ba58smith