SignalK/SensESP #427: cannot get a solid compile

PhilRWA, 2021-07-05:
Hi guys, total idiot here so please be gentle. I'vebeen trying to get the onewire temp sensor example to work on my d1 mini. I've so far had to use the PWM Range solution listed in a different issue, and select upload port manually, and include the SensESP library manually in the platform io file to get build to work. That has uploaded successfully, but then serial monitor gives me this:

␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/networking
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/sk
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/wifi_reboot_watchdog
␛[0m

Exception (9):
epc1=0x40205269 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff4dcd depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 0190
3ffffeb0:  001c001f 00000000 3fff4d44 4020525c  
3ffffec0:  00000000 00000000 00000000 40211154  
3ffffed0:  3fff4c04 3fffff30 3fff3098 40211260  
3ffffee0:  00000000 00000000 3fff3074 3fffff3c  
3ffffef0:  3ffe91cc 00000000 3fff3074 3fff2038  
3fffff00:  3fffdad0 402011a2 00000020 3fff2038  
3fffff10:  3fffdad0 3fff3074 3fff4d44 4020535c  
3fffff20:  3fffdad0 3fff3074 3ffe8c73 402011ab  
3fffff30:  00000000 feefeffe 00000000 00000000  
3fffff40:  feefeffe 00000000 00000000 feefeffe  
3fffff50:  00000000 00000000 00505345 00000000  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe 3fff2038  
3fffff80:  3fffdad0 00000000 3fff2024 401000ed  
3fffff90:  feefeffe feefeffe feefeffe 40201850  
3fffffa0:  feefeffe feefeffe feefeffe 402124f8  
3fffffb0:  feefeffe feefeffe 3ffe8680 4010045d  
<<<stack<<<

Anyone able to shed some light on this one? I'm at my wits end being like a say a total newbie to all this.

ba58smith, 2021-07-14:
@PhilRWA , can we take this to the #sensors channel on Slack? Since so many other users ARE able to compile successfully, I don't know that it's a bug in SensESP. More likely it's something with your configuration, which we can explore better on Slack.

JohnySeven, 2021-07-18:
@PhilRWA this issue isn't compile time related, but it's runtime exception that has occured. In order to find out what's wrong I recommend using esp stack trace decoder. It's little handy utility, that is able to decode call stack that is printed when ESP crashes into something meaningful. 
It's here: https://github.com/littleyoda/EspStackTraceDecoder
Let me know if you have any troubles getting it to work - it's fairly simple to run it and instructions are OK.
When you decode the call stack, just post the call stack here and then we can help more.

jasonhector, 2021-08-04:
I have travelled down a similar road to @PhilRWA attempting to use multiple ESP8266 units [nodemcu, ESP07, ESP12E] within Platformio. 
In itself there is a challenge to get the sample main.cpp to compile successfully using the default platformio.ini file for ESP8266 units. 

SPIFFS warning occur during build, and I suspect that the issue lies there in that the storage is not initiated correctly with the appropriate library, even through the build succeeds. I shall post the warning logs a bit later.

 I have ordered a few ESP32 units to test on as none of the ESP8266 units tested above either does not compile or contains a storage issue.

jasonhector, 2021-08-04:
Here is the compile warning on SPIFFS:

Compiling .pio/build/esp12e/lib01c/ESP Async WebServer/WebAuthentication.cpp.o
Compiling .pio/build/esp12e/lib01c/ESP Async WebServer/WebHandlers.cpp.o
In file included from .pio/libdeps/esp12e/ESP Async WebServer/src/SPIFFSEditor.cpp:1:
.pio/libdeps/esp12e/ESP Async WebServer/src/SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   16 |     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
      |                                                                                                     ^~~~~~
In file included from .pio/libdeps/esp12e/ESP Async WebServer/src/ESPAsyncWebServer.h:27,
                 from .pio/libdeps/esp12e/ESP Async WebServer/src/SPIFFSEditor.h:3,
                 from .pio/libdeps/esp12e/ESP Async WebServer/src/SPIFFSEditor.cpp:1:
/Users/nubbs/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |               ^~~~~~
Compiling .pio/build/esp12e/lib01c/ESP Async WebServer/WebRequest.cpp.o
Compiling .pio/build/esp12e/lib01c/ESP Async WebServer/WebResponses.cpp.o

Below the monitor output:

(E) (setup_wifi_manager) Failed to connect to wifi and config timed out. Restarting...
␛[0m
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000c49e0
~ld
(I) (SetupSerialDebug) 
␛[0m(I) Serial debug enabled
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/networking
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/sk
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/wifi_reboot_watchdog
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/analog_in
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/linear
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (enable) Connecting Signal K source SensESP.systemhz
␛[0m(I) (enable) Connecting Signal K source SensESP.uptime
␛[0m(I) (enable) Connecting Signal K source SensESP.freemem
␛[0m(I) (enable) Connecting Signal K source SensESP.ipaddr
␛[0m(I) (enable) Connecting Signal K source SensESP.wifisignal
␛[0m(I) (enable) Connecting Signal K source environment.indoor.illuminance
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (setup_discovery) mDNS responder started at SensESP
␛[0m*WM: Adding parameter
*WM: hostname
*WM: 
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
[... 81 lines cut ...]
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/wifi_reboot_watchdog
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/analog_in
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/linear
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (enable) Connecting Signal K source SensESP.systemhz
␛[0m(I) (enable) Connecting Signal K source SensESP.uptime
␛[0m(I) (enable) Connecting Signal K source SensESP.freemem
␛[0m(I) (enable) Connecting Signal K source SensESP.ipaddr
␛[0m(I) (enable) Connecting Signal K source SensESP.wifisignal
␛[0m(I) (enable) Connecting Signal K source environment.indoor.illuminance
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (setup_discovery) mDNS responder started at SensESP
␛[0m*WM: Adding parameter
*WM: hostname
*WM: 
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
*WM: 
*WM: Configuring access point... `

ruedigerO, 2021-09-01:
I run into the same issue. In my case it's related to the usage of Dallas DS18B20 temperature sensor.
I used the example from https://github.com/SignalK/SensESP/tree/master/examples/onewire_temperature.
Uploaded to an ESP32 and works fine!
Compiled and uploaded to an ESP8266 (D1_mini) results in a restart exception 9 loop.
Maybe it's related to the change to oneWireNG? In the past (March 2021) it was working.

A second test with the bme280_example works fine on both ESPs.

```
Output OneWire test:
␛[0m(I) Serial debug enabled
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/wifi_reboot_watchdog
␛[0m
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x4020526d epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff4d75 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 0190
3ffffeb0:  001c001f 00000000 3fff4d04 40205260  
3ffffec0:  00000000 00000000 00000000 40211154  
3ffffed0:  3fff3d24 3fffff30 3fff3018 40211260  
3ffffee0:  00000000 00000000 3fff2ff4 3fffff3c  
3ffffef0:  3ffe91cc 00000000 3fff2ff4 3fff1fb8  
3fffff00:  3fffdad0 402011a3 00000020 3fff1fb8  
3fffff10:  3fffdad0 3fff2ff4 3fff4d04 40205360  
3fffff20:  3fffdad0 3fff2ff4 3ffe8c73 402011ac  
3fffff30:  00000000 feefeffe 00000000 00000000  
3fffff40:  feefeffe 00000000 00000000 feefeffe  
3fffff50:  00000000 00000000 00505345 00000000  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe 3fff1fb8  
3fffff80:  3fffdad0 00000000 3fff1fa4 401000ed  
3fffff90:  feefeffe feefeffe feefeffe 40201854  
3fffffa0:  feefeffe feefeffe feefeffe 402124f8  
3fffffb0:  feefeffe feefeffe 3ffe8680 4010045d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000c6ea0
~ld
```

ba58smith, 2021-09-01:
I started having the same problem a few days ago when I needed to replace a fried Wemos D1 Mini. I think the problem is the OneWireNG library (and the related DallasTemperature library) that were replaced back on April 5, in order to make 1-Wire work (or work better?) with ESP32. I'm in the middle of reverting to the old libraries on my computer to see if the problem goes away.

ba58smith, 2021-09-01:
Yep, I think that's it. But I don't know why - the OneWireNG library repo says it's been tested with ESP8266. I'm going to try to track down the problem. @mairas, if you have any thoughts, I'd love to hear them.

ba58smith, 2021-09-01:
@ruedigerO or someone else: can you test this?
Go to .pio/libdeps/d1_mini/OneWireNg/src/OneWire.h
At line 69, remove `(_ow_buf)`, so it looks like this: `_ow = new OneWireNg_CurrentPlatform(`
Build it and test it.

I did that, and it now runs and doesn't crash. However, I'm away from all of my equipment except for an ESP8266, so I can't tell if it actually solved the problem, or just pushed it down the road, to when real sensors are actually connected.

ruedigerO, 2021-09-01:
@ba58smith
Yes I can confirm it works with this change!

Thanks for the quick workaround.
BTW: do you add a change request for this?

ba58smith, 2021-09-01:
To be clear, @ruedigerO , you've used it to send temperature data all the way to the Signal K Server? If so - excellent! I've already created an issue on the OneWireNg repo about it: https://github.com/pstolarz/OneWireNg/issues/15, so hopefully @pstolarz can look into it.

ba58smith, 2021-09-02:
@ruedigerO , as you may have seen on the Slack conversation about this, the bug has been fixed in the OneWireNg repo. The fix is a little more elegant than the hack I gave you above, but the end result is the same. https://github.com/pstolarz/OneWireNg/blob/master/src/OneWire.h#L158
Until he pushes a new version that PlatformIO will pick up automatically, you can copy the entire contents of that file in place of the entire contents of your `.pio/libdeps/d1_mini/OneWireNg/src/OneWire.h` file.
I'm going to close this issue.

closed 2021-09-02 by ba58smith

pstolarz, 2021-09-03:
Just a heads up. The new version of OneWireNg (0.8.3) has been just released. It'll take some time to synch with lib-repos (Arduino, PlatformIO) though.