SignalK/SensESP #428: ESP8266 no connection after WiFi setup

loefkvist, 2021-08-30:
My second attempt at creating a sensor using SensESP simply will not succeed.

I made a working sensor earlier (although some intermittent wifi-connection issues were present also then).
After updating to latest PlatformIO-IDE and hence new library versions it simply will not work.

As a test I created a new sensor, added the PWM-range define that apparently was removed for some reason, compiled using the analog_input.cpp example with no changes whatsoever and is getting the following each and every time I try to configure the ESP:

```
*WM: Scan done
*WM: DUP AP: loefkvist
*WM: DUP AP: loefkvist
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: loefkvist
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: loefkvist
*WM: DUP AP: loefkvist
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Sending Captive Portal
*WM: Sent...
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: loefkvist
*WM: DUP AP: loefkvist
*WM: Handle not found
*WM: Handle wifi
*WM: Scan done
*WM: Sent config page
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: loefkvist
*WM: DUP AP: loefkvist
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: Handle not found
*WM: Request redirected to captive portal
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: loefkvist
*WM: DUP AP: loefkvist
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: SensESP_test
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Handle not found
*WM: Request redirected to captive portal
(I) (wifi_station_connected) Connected to wifi, SSID: loefkvist (signal: -62)
␛[0m(I) (wifi_station_connected) IP address of Device: (IP unset)
␛[0m*WM: Connection result:
*WM: 3
(I) (setup_wifi_manager) Connected to wifi,
␛[0m(I) (setup_wifi_manager) IP address of Device: (IP unset)
␛[0m(I) (setup_wifi_manager) Got new custom hostname: SensESP_test
␛[0m(I) (setup_wifi_manager) Got new SSID and password: loefkvist
␛[0m(W) (setup_wifi_manager) Restarting in 500ms
␛[0m(D) (enable_all) Enabling all required sensors and transforms
␛[0m(I) (enable) All sensors and transforms enabled
␛[0m(I) (connect) WSClient websocket connect attempt (state=0)
␛[0m(D) (connect) Initiating websocket connection with server...
␛[0m(E) (connect) No Signal K server found in network when using mDNS service!
␛[0m(I) (wifi_station_disconnected) Disconnected from wifi.
␛[0m
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000c4ad0
~ld
(I) (SetupSerialDebug) 
␛[0m(I) Serial debug enabled
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified: 
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/sk
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /system/wifi_reboot_watchdog
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/analog_in
␛[0m(I) (load_configuration) Not loading configuration: file does not exist: /indoor_illuminance/linear
␛[0m(I) (load_configuration) Not loading configuration: no config_path specified:
␛[0m(I) (enable) Connecting Signal K source SensESP_test.systemhz
␛[0m(I) (enable) Connecting Signal K source SensESP_test.uptime
␛[0m(I) (enable) Connecting Signal K source SensESP_test.freemem
␛[0m(I) (enable) Connecting Signal K source SensESP_test.ipaddr
␛[0m(I) (enable) Connecting Signal K source SensESP_test.wifisignal
␛[0m(I) (enable) Connecting Signal K source environment.indoor.illuminance
␛[0m(I) (enable) Enabling subsystems
␛[0m(I) (setup_discovery) mDNS responder started at SensESP_test
␛[0m*WM: Adding parameter
*WM: hostname
*WM:
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result:
*WM: 0
*WM: SET AP STA
*WM:
*WM: Configuring access point...
*WM: Configure SensESP_test
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: DUP AP: loefkvist
```
...and it's back as an AP for configuring.
Yes, I am using the correct WIFI-credentials ;) 

Device is a Wemos D1 mini.
Result is the same when using an ESP12e.

I'm consistently seeing `IP address of Device: (IP unset)` and it seems the device is never getting an IP-address. 

Any help will be appreciated.

loefkvist, 2021-08-31:
Oh...
Actually now realize that there's a reboot in there caused by a SW WD reset. 
Is there a busy-loop somewhere causing this?

mairas, 2021-09-01:
Could you share a simple test program that exhibits this behavior?

Also, you might want to try whether you'll get an IP address when connecting a different access point (a phone hotspot would also do).

Finally (although probably not related to this issue), I strongly recommend to use an ESP32 for your SensESP projects. The price difference is negligible, the hardware much more capable, and most importantly, we don't at the moment have any capable developers who would have time to maintain the ESP8266 side of things.

ruedigerO, 2021-09-02:
@mairas: I've see the same behavior. It happens only on ESP8266 and SensESP release version (1.0.5).
ESP32 is working well.
With the 'master' version (1.0.6-alpha) it's also working on ESP8266 and the ESP will get an IP address.

loefkvist, 2021-09-02:
> Could you share a simple test program that exhibits this behavior?
>
> Also, you might want to try whether you'll get an IP address when connecting a different access point (a phone hotspot would also [...]

It is seriously just a new, clean project with the standard platformio.ini and the analog_input example.
No changes to anything.

As for using an ESP32; I have a pile of ESP12e modules laying around. Would be a shame to just bin them.

loefkvist, 2021-09-02:
> @mairas: I've see the same behavior. It happens only on ESP8266 and SensESP release version (1.0.5).
> ESP32 is working well.
> With the 'master' version (1.0.6-alpha) it's also working on ESP8266 and the ESP will get an IP address.

Nice - I'll try 1.0.6-alpha out ;)
Thnx.

loefkvist, 2021-09-03:
> @mairas: I've see the same behavior. It happens only on ESP8266 and SensESP release version (1.0.5).
> ESP32 is working well.
> With the 'master' version (1.0.6-alpha) it's also working on ESP8266 and the ESP will get an IP address.

Ok, tried the master-branch.....
Now getting an IP-address but the device still reboots:
`␛[0m(I) (setup_wifi_manager) IP address of Device: 192.168.8.215
␛[0m(I) (setup_wifi_manager) Got new custom hostname: SensESP_test
␛[0m(I) (setup_wifi_manager) Got new SSID and password: loefkvist
␛[0m(W) (setup_wifi_manager) Restarting in 500ms
␛[0m(D) (enable_all) Enabling all required sensors and transforms
␛[0m(I) (enable) All sensors and transforms enabled
␛[0m(I) (connect) WSClient websocket connect attempt (state=0)
␛[0m(D) (connect) Initiating websocket connection with server...
␛[0m(E) (connect) No Signal K server found in network when using mDNS service!
␛[0m(I) (wifi_station_disconnected) Disconnected from wifi.
␛[0m
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000c4a50
~ld
(I) (SetupSerialDebug) 
␛[0m(I) Serial debug enabled`

I'll try to look into this further.

patrol420, 2021-09-05:
i had the same issue and i ended up solving it how it ever was about 4 months i cant remember. some people on slack helped me ill see if i can find the thread

patrol420, 2021-09-05:
goto your platformio.ini file and below the [ENV] you have 

; This one is the "Release" version - the branch called "latest".
 ; SignalK/SensESP

; This one has all merged PR's, but is not yet an official "release" version.
; It is the branch called "master". This version is unstable and under continuous
; development and will break without warning. Use it only if you want to
; participate in SensESP development or need to test some yet unreleased feature.
https://github.com/SignalK/SensESP

comment the top one and uncomment the web address to github

loefkvist, 2021-09-06:
@patrol420 

Thanks.
Hardcoding the WIFI-credentials seem to make it work.

grhitch, 2021-11-06:
Having the same trouble. Tried the "master" version just in case, but no luck.
Could it be an upstream issue caused by the changes referred to in this issue?

[https://github.com/esp8266/Arduino/pull/7902](https://github.com/esp8266/Arduino/pull/7902)

This issue in EspAsyncWiFiManager seems to concur:
[https://github.com/alanswx/ESPAsyncWiFiManager/issues/83](https://github.com/alanswx/ESPAsyncWiFiManager/issues/83)

If I read it correctly, the default behaviour of writing the WiFi credentials to flash on successful connect was changed in the platform core. When the esp8266 reboots it forgets the connection information.

grhitch, 2021-11-06:
Confirming the above. In networking.cpp, I added the line:
"WiFi.persistent(true);"
before the line
"wifi_manager = new AsyncWiFiManager(server, dns);"
I can now configure the ESP device with the portal and the settings are remembered.

I don't know enough to know if this is the best or complete solution. If you believe that this is the right approach, I am happy to create a PR.

mairas, 2021-11-06:
I haven't tested it, but skimming through the linked material does also lead me to believe that your proposal is the right way to deal with this. If you want to do a PR, I'd be happy to merge it!

PR #463 (2021-11-07): WiFi Credential Persistence Across Reboot
Fixes #428.
Refer also esp8266/Arduino#7902 and alanswx/ESPAsyncWiFiManager#83
closed 2021-11-07 by mairas