SignalK/SensESP #450: WiFi as option?

ruedigerO, 2021-11-05:
Is there a way / an option to make the WiFi connect optional?

I would like to achieve the following:
SensESP connected to some sensors and a local (small) display
if signalk server / Wifi is up -> connect to it, get sensor data and send it to the local display
else get sensor data and send it to the display, after a while (1-2 minutes) try to connect WiFi again.

Right now the ESP is restarting (after 3 minutes) if there is no WiFi.
It would be nice to get the sensor part working without WiFi.

Regards

mairas, 2021-11-05:
Working on it! The repo has a `v2-dev` branch that includes a lot of work towards the next release. One major feature is the ability to create "minimal" SensESP apps in which you can pick just the features you need. You can have a look at `examples/minimal_app.cpp` in that branch.

Beware, though: there has been recently a huge merge-fest of changes against that branch, in which I've tried to split my monolithic experimental development branch into bite-size PRs. That work is almost complete but there are a couple of PRs left, and probably some more work after that. At the moment I think it's likely the branch isn't in a usable state because I haven't tested every single PR on real hardware, and it's quite likely I've messed up something. But after this weekend it might be a different story.

ba58smith, 2021-11-05:
Based on the work you've already showed us (in your PR's) @mairas, I don't recall seeing anything that would make wifi connection optional in the way that @ruedigerO is asking about. i.e., connect to wifi if it's available, but if it's not, keep going and send data to the local display. Then periodically check for wifi, and connect if / when it becomes available.

Seems like the wifi module would need a modification to allow such behaviour, no?

mairas, 2021-11-05:
Ah. Fair enough. I was only considering the heading literally: to make the wifi truly an optional component. But yes, the three-minute reboot timeout is a bit harsh in such use cases. But then again, in the basic use case in which we expect WiFi to be available, I don't think it's all that bad. A builder flag to disable the 3-minute kill switch could be added, perhaps.

ruedigerO, 2021-11-06:
The 3-minute kill switch is not the problem. Problem is: all sensors are not enabled until a WiFi connect was successful.
So, to show data on a local display a WiFi connection must be established - which makes the local display useless.
(data can be shown in SK)
My understanding was SensESP = Sensor_for_ESP not Sensor_for_SignalK
Correct me if I am wrong!

ba58smith, 2021-11-07:
Well, you are wrong with the current implementation of SensESP. It is really only for sending date to Signal K, but you can also send data to a local display. What @mairas is working on now will change that.

pa1pdr, 2022-01-31:
I'm extremely interested in an 'optionality to WiFi' too. Without it Wifi becomes the SPoF (single point of failure) for all ESP projects built with SensESP, which would make it useless in my case. N2K communication is my primary target (using Matti Arias' excellent ESP32 HatLab board). It would be nice if one could measure in a loop and only connect to SK if the wifi is available and skip it otherwise. Any thoughts?

ckcrs, 2022-02-07:
+1. Considering testing with SensESP away from SignalK&wifi would be a beneficial, thanks

mairas, 2022-02-07:
This is definitely an issue that I want to work on in the near future. It should be somewhat more doable now that there's just the ESP32 platform to consider.

The approach I have in my mind is to identify the blocking calls in `networking.cpp` (at least all HTTP client calls) and then make them asynchronous in one way or another; probably by creating FreeRTOS tasks for them.

pa1pdr, 2022-02-07:
I've found a nice way around it: I use a separate ESP32, connected to the engine (and firing up when the engine contact is closed). It gathers info using OneWIre tem sensors and i2c INA226 electrical power from generator and engine and transmits that periodically via the ESPNow protocol. This is non-SensESP, just plain C++ and Arduino libs.
A second HatLab ESP32 referenced before (w. SenseESP 2.1)  is fired up whenever N2K is powered, connects to the WiFi and SignalK, which listens to ESPNow telegrams and dispatches the relevant values on N2K and SignalK. 

The concept works quite well and gives me independence from WiFi or N2k to  be present, which also opens the door for continuous measurements like battery capacity (with a separate ESP32 and INA226).

This increases the amount of ESP's to feed (12V & $), but also overal robustness.

PR #556 (2022-02-15): Async WSClient
The SK websocket token testing and connection establishing has been heavily blocking, stopping the program execution until the connection has been established. This behavior prevents for example seamless simultaneous N2K and SK output.

This PR moves the SK websocket operations to a separate FreeRTOS task. Concurrency between the main and WSClient tasks is ensured with mutexes and task queues.

Fixes #450.
closed 2022-02-15 by mairas