SignalK/SensESP #452: Update the JSON Editor

mairas, 2021-11-06:
The embedded [JSON Editor](https://github.com/json-editor/json-editor) code could be updated. And while at it, apply some nicer theme to it as well!

JohnySeven, 2021-11-07:
I was thinking about adding bootstrap into bundle and making the sice more nice with AJAX and some tiny Javascript framework, ideally SPA would be nice! All website resources can be GZIPed first to save flash and serve them as GZIPed via HTTP.

mairas, 2021-11-07:
Sounds pretty great. I'm not much of a frontend developer, so my effort in that front tends to be bare minimum. 😁 As long as the flash use stays somewhat modest and everything works offline (without resources pulled from the internet), I think we're good!

JohnySeven, 2021-11-07:
And also I would add ability to add commands from builder when using HTTP server - for example if you want to trigger some calibration process or reset some values without loosing Wifi config it would be good to have.

JohnySeven, 2021-11-07:
Well I have this idea in my mind for long time...not making promises, but I would like to contribute with web UI :-)

mairas, 2021-11-07:
I wouldn't use the builder for that. Instead, how about some kind of inversion of control class that would allow the user to register new configuration objects. Like the resettables etc.

mairas, 2021-12-16:
Not really required for the 2.0 release (because the change shouldn't be backwards-incompatible). Removing the label.