SignalK/SensESP #454: Add a default password to the WiFi captive portal

mairas, 2021-11-06:
If a SensESP device can't connect to the WiFi, it automatically sets up its own configuration access point. This access point has no password and makes it trivial to hijack a device in a harbor or anchorage, for example. Set a default password to the captive portal.

JohnySeven, 2021-11-07:
Just to clarify: Is this only in case the WIFI configuration isn't hardcoded?

mairas, 2021-11-07:
> Just to clarify: Is this only in case the WIFI configuration isn't hardcoded?

Yes. The idea is that, let's say, you're in a busy marina and you power on your electronics but the WiFi router is offline, then any SensESP devices (that don't have hardcoded WiFi config) would enable the config AP, allowing any neighbor to hijack the device.

mairas, 2021-12-16:
Fixed in #481.

PR #481 (2021-12-08): Set a default password for the WiFi Manager configuration access point
WiFi Manager configuration access point is started if the WiFi configuration is not hardcoded and the device is unable to connect to the WiFi. Previously, this AP didn't have a password, allowing anyone nearby to hijack the device.

This PR adds a default password ("thisisfine") for WiFiManager. A builder method for changing the default password has also been implemented.

Fixes #454.
closed 2021-12-16 by mairas