SignalK/SensESP #455: Reimplement previously removed "standard sensors"

mairas, 2021-11-06:
There are ~four~ five system information sensors, previously called standard sensors, that were enabled by default and could be triggered on or off using SensESPAppBuilder:

- SystemHz
- FreeMem
- Uptime
- IPAddrDev
- WifiSignal

SensESPAppBuilder needs to have convenience methods to enable these:

- `enable_system_hz_sensor()`
- `enable_free_mem_sensor()`
- `enable_uptime_sensor()`
- `enable_ip_address_sensor()`
- `enable_wifi_signal_sensor()`

And additionally, one convenience method to enable all of these. My proposal for the name:

- `enable_system_info_sensors()`

ba58smith, 2021-11-06:
I like `system_info_sensors`.

JohnySeven, 2021-11-07:
Like it! Library user have it all under control!

mairas, 2021-11-29:
Fixed in #472.

closed 2021-11-29 by mairas