SignalK/SensESP #457: Remove ESP8266 code paths

mairas, 2021-11-06:
Version 2 will no longer support ESP8266; the conditional code paths should be removed.

ba58smith, 2021-11-06:
Bravo!

JohnySeven, 2021-11-07:
Great! But I would leave at least one def something like

```
#if ESP8266
#error "ESP8266 isn't supported by SensESP v2"
#endif
```
Just to be clear when somebody forgot to specify v1 in platformio.ini, this would stop him right there.

mairas, 2021-11-07:
> Great! But I would leave at least one def something like
> 
> ```
> #if ESP8266
> #error "ESP8266 isn't supported by SensESP v2"
> #endif
> ```
> 
> Just to be clear when somebody forgot to specify v1 in platformio.ini, this would stop him right there.

This is the way to go! And furthermore, the error message should link to instructions on how to peg SensESP to v1 to keep existing code from breaking.

ba58smith, 2021-11-07:
@mairas, when you're almost finished with all the V2 changes, let me know, and I'll start working on redoing the documentation. (Yee ha! My favorite thing!)

mairas, 2021-11-07:
This time, I'd like to participate in the documentation work as well!

ba58smith, 2021-11-07:
Be my guest! Feel free to write any of it that you'd like, or to direct how you'd like for it to be organized, and I can handle the details.

mairas, 2021-12-04:
Fixed in #478. Why didn't that PR close this issue automatically?

PR #478 (2021-11-30): Remove ESP8266 support
Remove ESP8266 specific code paths, comments and dependencies.

Fixes #457.
commit 718ced1e (2021-11-30): Add a guard statement for ESP8266

Fixes #457.
closed 2021-12-04 by mairas