SignalK/SensESP #473: ADS1115 sensor code does not work when one of the ADC inputs is grounded.

RBerliner, 2021-11-25:
The ADS1115 ADC outputs a 2's complement 16 bit binary integer on conversion with +MAX FS as 7FFF h and -MAX FS as 8000 h. If the ADC input AINx is grounded, the ADC can still output a small negative value because of noise or internal offsets. The ADS1x15 sensor code does not handle this case, treating the ADC output as a unsigned integer so that the grounded ADC voltage output appears as nearly 2x the MAX FS range.

mairas, 2021-11-27:
Thanks for the bug report. But darn it, I didn't notice Butch had already stepped up. I just created PR #474 to fix the issue. Sorry for barging in!

ba58smith, 2021-11-27:
@mairas - don't be silly. I assigned it to myself only so I wouldn't forget to fix it once I've stopped traveling around so much. I didn't have any particular fondness for it. :-)

rhanumaiah, 2021-11-27:
I got one of these and I am experiencing this issue. The hardware is on my bench, so if you need me to assist in any way. I will. Not that I'm an expert at all. This is my 1st sensor, and it's been a challenge. Its on a esp8266 and an add1115. Butch pointed me to this thread.

ba58smith, 2021-11-27:
> I got one of these and I am experiencing this issue. The hardware is on my bench, so if you need me to assist in any way. I will. Not that I'm an expert at all. This is my 1st sensor, and it's been a challenge. Its on a esp8266 and an add1115. Butch pointed me to this thread.

@rhanumaiah - this has been fixed, but not tested. If you look at https://github.com/SignalK/SensESP/pull/474/files, you can see what the change was (the code in pink is what it was, the code in green is what it was changed to). Make that same change on your computer, to the file src/sensors/ads1x15.h, then rebuild it and re-upload it, and see if it solves the problem. Report your results back here, please.

rhanumaiah, 2021-11-27:
> @rhanumaiah - this has been fixed, but not tested. If you look at https://github.com/SignalK/SensESP/pull/474/files, you can see what the change was (the code in pink is what it was, the code in [...]

Seem to still be bounding around now between 0 and (-0.00125 to - 000625) . Made this change in .pio/d1-mini/SensESP/src/sensors/ads1x15.h in line 70.  Made the change saved the file, ran a clean, then built, then updated OTA. Been having issues with settings, had to erase the flash the last time to fix an issue. I’ll try that.

ba58smith, 2021-11-28:
Can you provide a known, solid voltage and get a solid, not-bouncing-around output? In other words, does the bouncing around happen only when your voltage input is 0.0 (GND)?

RBerliner, 2021-11-28:
Hi Brian:

I am setup and testing the fix now. It took a little while because the
"production" device I was working on had some wiring bug so I am back to
the breadboard.

With this setup, when the input is grounded, I observe a solid response
that is a few ADC counts below zero i.e. 65534. This will bounce around a
few counts in the LSD. If the wiring has greater noise, you can see the
calculated voltage switch from 0 to something near 12.28 volts.

With the input tied to 3.3V , I get a solid 3.3026 v bouncing around a
little in the LSD.

I am now ready to apply the fix. I have cloned the latest version of
SensESP from github. Forgive me but I only do simple things on git. Should
I do a pull the FIX version which I could merge and then point libdeps to
my local copy of SensESP or do I simply change platformio.ini
changing
libdeps = SignalK/SensESP
 to
libdeps/SensESP:c4bab8d
<https://github.com/SignalK/SensESP/pull/474/commits/c4bab8dad96651cb971e6c3a2a2a080ce6d0cf23>

to apply the fix to my build?

Thanks

Ron

rhanumaiah, 2021-11-28:
> Can you provide a known, solid voltage and get a solid, not-bouncing-around output? In other words, does the bouncing around happen only when your voltage input is 0.0 (GND)? 

Seems to bounce around the top end too. 12.3665, 12.367, and 12.36563. So it bounces around at the ground and a constant voltage supply.

ba58smith, 2021-11-28:
@RBerliner , the fix is literally on character. Just open the file in PlatformIO (src/sensors/ads1x15.h) and change line 70 from `uint16_t raw_value_ = 0;` to `int16_t raw_value_ = 0;`. (Remove the `u` at the beginning.) Save, build, upload, test.

ba58smith, 2021-11-28:
@rhanumaiah - You're talking abouot a "bounce" of less than .01 percent. I don't think you'll ever get anything better than that on components that cost a few dollars. By the time you round your output to 1 or 2 decimals, you'll never see this tiny fluctuation. And besides - this isn't a software problem, it's hardware / wiring / noise / etc.

After applying the fix (changing `uint16_t` to `int16_t`), please test at GND, and if you're getting 0.0 plus or minus 0.01, then I think we can safely say the fix works.

RBerliner, 2021-11-28:
I confirm that the fix works. With the ADC input grounded, the raw counts
from the ADC and the calculated voltage are negative (In the case of this
simple breadboard setup ADC_counts = -2 and electrical.rawVolts.voltage =
-0.000375).

But that method would not work when the original and new source were more
different.

ba58smith, 2021-11-28:
> But that method would not work when the original and new source were more different.

Sorry, I don't understand what you mean by this comment, @RBerliner .

rhanumaiah, 2021-11-28:
> @rhanumaiah - You're talking abouot a "bounce" of less than .01 percent. I don't think you'll ever get anything better than that on components that cost a few dollars. By the time you round your [...]

Totally agree. how do I round the output to two decimals points within the code? Thanks

ba58smith, 2021-11-28:
@rhanumaiah - you CAN do it in the code with the SensESP MovingAverage transform. However, with such small fluctuations, I don't think you need to. When the number is finally displayed for human consumption (in Instrument Panel, or Wilhelm SK, etc.), if you display only 1 or 2 decimals, those fluctuations will simply disappear.

RBerliner, 2021-11-29:
@Butch:

What I was trying to say was if the "fix" was more complicated, it would be
onerous to make the corrections to SensESP by hand.

I thought at first that the head of the GitHub repo might have the
corrections but they are not there. Then I looked for another branch -
something like "BugFix" but I did not see that either. I was trying to find
a branch that *I could pull into my local SensESP, merge and then point
platformio.ini to my local copy or SensESP.. *

*Is that how you do it?*

*Where is the branch?*

*Thanks *

*Ron*

RBerliner, 2021-11-29:
@Butch:
BTW. Have I approved the check of the fix correctly? Somehow I suspect that
there is a more automated way to do it than to reply in the comments chain
of the Issue.

ba58smith, 2021-11-29:
@RBerliner , the fix has not been merged yet - so there is no version that you can point to. That's why I gave the advice to simply make the change that's in the Pull Request directly to your copy of the code. If you will post you comment about testing the fix to the Pull Request, that will give @mairas the confidence that he can merge it. Thanks. https://github.com/SignalK/SensESP/pull/474

mairas, 2021-11-29:
@RBerliner you can indeed pull changes from PRs for testing, and that definitely is preferred to manually replicating the changes in a PR. At the top-right corner of the PR page, there's a "<> Code" Button. Clicking that will give you instructions on how to check out the PR locally for inspection. You need the GitHub command line tool for that, though (or at least it will make your life a lot easier).

And as for "formal" approvals, I think you need to do a code review in the PR to access that (by clicking on the "Files changed" tab and then the green "Review changes" button. But usually it's enough to just comment on the PR.

But anyway, thanks for verifying the fix. I'll merge it.

PR #474 (2021-11-29): ADS1115 output should be signed
One-character bugfixes are the best.

I didn't have suitable hardware at hand so I didn't actually test the change. Let's summon @RBerliner for that!

Fixes #473.
closed 2021-11-29 by mairas