SignalK/SensESP #476: GIT seesm to crash the Open Plotter & SensESP image

Latest-Trick, 2021-11-27:
Good afternoon team,
I've recently discovered SensESP as a solution to my boats IoT. It looks like it's going to do everything I need and from the initial setup was working great reading data from a remote ESP32 (Firebeetle from DFRobot) and sharing across the signal K network......
BUT
When I reboot the RPi, it boots to a console window and I can't work out how to get back to the Open Plotter GUI.
After rebuilding the system image a few times (rebooting at each stage of the setup) I discovered the error state happens after the installation of GIT (https://git-scm.com/book/en/v2/Getting-Started-Installing-Git). The system still works ok after the installation, it's just when it's rebooted, sending me into the console window.
I'm sure I'll need to give more info on the error state so please be gentle....
Current system is RPi4(4GB) running from a 64Gb SD card with a standard OpenPlotter image, dAISy & Uputronics GPS HATs 

Any thoughts on what I'm doing wrong? Thanks in advance,

ba58smith, 2021-11-27:
I'm 99.9% sure this isn't a SensESP issue. Please post this in the #support channel on the Signal K Slack. http://slack-invite.signalk.org/

Latest-Trick, 2021-11-27:
Thanks for the quick reply & agree suspect it's not SensESP but thought it was a good place to start. 
I'll try the Signal K slack.

closed 2021-11-27 by Latest-Trick